github0null/eide #6: 生成.hex文件时出现错误

mengzhuozhuo, 2019-11-18:
*** ERROR L257: UNKNOWN

 ERROR  生成失败(build Failed) : Link Error

github0null, 2019-11-18:
有完整的命令行截图吗

mengzhuozhuo, 2019-11-18:
PS C:\Users\asus\Desktop\C51_programe\Target 1> & 'c:\vscode\extensions -s 'c:\Users\asus\Desktop\C51_programe\Target 1\.\src\Source Group 1' 
Group 1;c:\Users\asus\Desktop\C51_programe\Target 1\.\dependence\build-uild-in\STD;d:\software\Keil\C51\INC' -d 'c:\Users\asus\Desktop\C51_pro

[ INFO ] Source file list:

 WARNING  Not found C51 Startup File! File must be named like this: xxx

> C File:  c:\Users\asus\Desktop\C51_programe\Target 1\.\src\Source Gro

[ INFO ] --------------------Start Compile...--------------------

C51 COMPILER V9.60.0.0 - SN: T1UMC-G5KF4C
COPYRIGHT Copyright (C) 2012 - 2019 ARM Ltd and ARM Germany GmbH. All r

C51 COMPILATION COMPLETE.  0 WARNING(S),  0 ERROR(S)

--------------------------------------------------------

******************************************************************************
* RESTRICTED VERSION WITH 0800H BYTE CODE SIZE LIMIT; USED: 0001H BYTE ( 0%) *
******************************************************************************

Program Size: data=9.0 xdata=0 const=0 code=16
*** ERROR L257: UNKNOWN

 ERROR  生成失败(build Failed) : Link Error

mengzhuozhuo, 2019-11-18:
[image]

github0null, 2019-11-18:
看起来并不是插件本身的问题，插件调用的是 Keil 编译器，
帮你找到了一个 链接， 上面好像有 和你一样的问题， [here](https://community.cypress.com/thread/44542)

github0null, 2019-11-18:
可能是没有完全破解 Keil，还存在代码限制

github0null, 2019-11-18:
建议先导出 Keil 项目在keil 上编译一下，如果能过，那应该是我的构建工具有问题，

mengzhuozhuo, 2019-11-18:
keil上也不能编译，我再重装试一试。麻烦你了

mengzhuozhuo, 2019-11-18:
[image]
我发现导出的项目只要把这两个选项去掉就可以编译通过，我在vscode上改怎么做？

mengzhuozhuo, 2019-11-18:
[image]
我发现导出的项目只要把这两个选项取消就可以编译，在vscode上应该怎么做？

github0null, 2019-11-18:
这个插件只使用 LX51 做链接器， AX51做汇编器, 暂时没法更改，因为 LX51 功能更加强大

mengzhuozhuo, 2019-11-18:
那我改用其他版本的keil是不是就可以了？

github0null, 2019-11-18:
不知道，具体看你代码中写了什么，我已经编译过很多C51项目了，都是没问题的， 后续我将会加入一些可修改的编译参数到插件中，增强插件的功能，

> 如果可以的话，你可以把你出问题的工程打包发我，这样可以帮助我改进这个插件，让它变得更可靠，稳定，谢谢

mengzhuozhuo, 2019-11-18:
[test.zip](https://github.com/github0null/eide/files/3857927/test.zip)

我的代码什么也没写，应该是我操作有问题，你要是有空的话帮我看一下。谢谢

github0null, 2019-11-18:
> 在我的电脑上没问题的，其实也不用纠结这些， 实在不行，在keil 上编译也可以的
[image]

mengzhuozhuo, 2019-11-18:
嗯，麻烦你了

closed 2020-01-05 by github0null

tinyhugh, 2024-04-11:
我这边碰到和你一样的情况，后来发现是权限问题，如果你是使用注册机，可以选择Prof. Developers Kit (Plus)，留下此解决办法帮助后来人。