github0null/eide #7: ERROR  : Invalid bin dir !, [path] : c:\Keil

linlinlgh, 2019-11-19:
PS C:\Users\Administrator\Desktop\F4\F1\USER\ICAPTURE> & 'c:\Users\Administrator\.vscode\extensions\cl.eide-1.0.4\bin\ARM\ARM_Builder.exe' -b 'c:\Keil' 'c:\Users\Administrator\Desktop\F4\F1\USER\ICAPTURE\.\out\ICAPTURE.params'

github0null, 2019-11-19:
> 如果在编译前 Keil 的路径是无效的话，会发生此类错误，**`应该重新设置新的有效的 Keil 路径`**，
****
出现此错误，应该是你在命令行界面使用 ⇡ 键回到了上次的命令，并回车进行了编译； 正确的方法是点击编译按钮，这样在编译前插件会进行目录有效性的检测，从而阻止错误编译的发生，会给出警告提示
****
下图是类似的错误
****
[image]

linlinlgh, 2019-11-20:
我在上面设置keil路径后选择的keil安装目录下的TOOLS.INI，是这样吗？

github0null, 2019-11-20:
是的，如果设置错了会提示找不到编译工具

linlinlgh, 2019-11-20:
导入keil工程设置好后，还是会出现
Windows PowerShell
版权所有 (C) Microsoft Corporation。保留所有权利。

尝试新的跨平台 PowerShell https://aka.ms/pscore6

PS E:\test\USER\Target 1> & 'c:\Users\Administrator\.vscode\extensions\cl.eide-1.0.4\bin\ARM\ARM_Builder.exe' -b 'c:\Keil' 'e:\test\USER\Target 1\.\out\Target 1.params'l.eide-1.0.4\bin\ARM\ARM_Builder.exe' -b 'c:\Keil' 'e:\test\USER\Target 
1\.\out\Target 1.params'

 ERROR  参数错误 : Invalid bin dir !, [path] : c:\Keil

github0null, 2019-11-20:
这就奇怪了，只有在目录不存在的时候才会有这个错误，也许是在C盘下，命令行的权限不够把，原因目前还不清楚

linlinlgh, 2019-11-20:
好吧

lmhtz, 2019-11-25:
keil 版本是否有要求？ 无法设立C:\Keil_v5\TOOLS.INI路径，版本5+

github0null, 2019-11-25:
并没有版本要求，设置目录只是为了找到编译工具路径

magicsjw, 2019-12-10:
我也遇到同样问题，在windows 7 x64 操作系统上， 
[image]
在vscode里选择PowerShell时，报ERROR : Invalid bin dir !, [path] : ...
在vscode里选择PowerShell Core时，报这个错误截图：
[image]

github0null, 2019-12-10:
你好，如果 keil 安装在C盘下就会这样，因为构建工具权限不够，这是没法避免的

如果安装目录没问题，那就是编译工具路径没选对， 

而且应该使用Powershell 而不是Powershell core

magicsjw, 2019-12-10:
> 你好，如果 keil 安装在C盘下就会这样，因为构建工具权限不够，这是没法避免的
> 
> 如果安装目录没问题，那就是编译工具路径没选对，
> 
> 而且应该使用Powershell 而不是Powershell core

我的keil安装路径不是C盘，路径应该没问题，设置路径那里我把C51和ARM都选了D:\Keil\tools.ini
[image]

github0null, 2019-12-10:
ARM 的 tools.ini 路径 和 C51 的 是不同的，

> ARM：ARM 可执行文件 “UV4.exe” 的上一层目录里的 Tools.ini
> C51：C51 可执行文件 ”UV4.exe“ 的上一层目录里的 Tools.ini

**这个问题下一个版本会解决的**

linlinlgh, 2019-12-13:
最近忙其他去了，今天看了下，编译成功了，需要注意几个问题：1、导入c51工程是.uvproj，也就是要用Keil uVision4,导入arm工程是.uvprojx，也就是Keil uVision5的版本，2、选择设置keil的路径要区分是C51还是ARM的Tools.ini，3、keil软件一定不能安装在C盘，然后试了试代码提示，确实强大

closed 2020-01-05 by github0null