github0null/eide #10: 编译时的 shell 必须是 powershell，后续将会改进

github0null, 2019-12-05:


3110962976kuangnu, 2019-12-13:
在win7上使用powershell
编译时出现以下：

> 'powershell' 不是内部或外部命令，也不是可运行的程序
或批处理文件。

github0null, 2019-12-13:
检查你的电脑是否安装了 powershell，我记得 win7 是有的

3110962976kuangnu, 2019-12-13:
> 检查你的电脑是否安装了 powershell，我记得 win7 是有的

确实有，在开始菜单附件里可以找到

多谢您的回复  我还可以用win10

下面是我查到的我的win7上面的powershell的版本
> PS C:\Users\Administrator> $PSVersionTable

Name                           Value
----                           -----
CLRVersion                     2.0.50727.8784
BuildVersion                   6.1.7601.17514
PSVersion                      2.0
WSManStackVersion              2.0
PSCompatibleVersions           {1.0, 2.0}
SerializationVersion           1.1.0.1
PSRemotingProtocolVersion      2.1

github0null, 2019-12-14:
在 cmd 中运行 powershell 如果没有进入到 powershell，那可能是 powershell 没有被加入到环境变量中

closed 2020-03-15 by github0null