github0null/eide #11: 编译时候会找不到头文件

qihuanlu01, 2019-12-06:
"c:\Users\MicroLin\Documents\STM32\vSCODE\LED_test\.\src\Drivers#STM32F1xx_HAL_Driver\stm32f1xx_hal_rcc_ex.c", line 116: Error:  #268: declaration may not appear after executable statement in block
      FlagStatus       pwrclkchanged = RESET;
      ^
c:\Users\MicroLin\Documents\STM32\vSCODE\LED_test\.\src\Drivers#STM32F1xx_HAL_Driver\stm32f1xx_hal_rcc_ex.c: 0 warnings, 1 error

 ERROR  生成错误 : Unknown Error On:  c:\Keil_v5\ARM\ARMCC\bin\armcc.exe

在keil内编译没有问题。头文件加载不进来

github0null, 2019-12-06:
因为 keil 内是安装了 STM32 固件包的， 如果你知道头文件所在目录，可以在 **自定义依赖中** 添加 头文件目录， 当然你也可以 直接安装 keil 包， 可以在arm官网下载，安装完包之后，可以直接添加依赖，完成头文件目录的设置

qihuanlu01, 2019-12-10:
添加过了，还是不行。declaration may not appear after executable statement in block
FlagStatus pwrclkchanged = RESET;
^
还是这个问题，头文件应该有了，就是加载在.c文件后面。我是导入的keil项目，stm的固件包安装过了。
要是能直接打开原项目就好了，现在都有git了，觉得原项目备份没必要

github0null, 2019-12-10:
**1.** 出现这个错误应该是代码的问题，，编译工具是 Keil ARMCC，默认的语言标准是 C89， 变量声明不能在语句后， 你可以在编译选项->languageMode 中更改语言标准

**2.** 这样做的目的并不是为了备份原项目，Keil 项目的文件很分散，并且 keil 的源文件是 ANSI 编码的，在 vsCode 上是乱码，而且不便于 vscode-cpptools 进行语法分析，这样写代码的体验就和 Keil 一样惨不忍睹了，就偏离了原来的初衷， 因此本插件在导入时会创建新的项目而不是直接使用 Keil 的 .uvprojx 文件

github0null, 2019-12-10:
包含头文件是预编译命令，在编译前头文件就被直接嵌入了源文件，是不存在在源文件之后加载的，除非你的语句写在 #include 前面

github0null, 2019-12-10:
这些问题会改进的

qihuanlu01, 2019-12-11:
是编译选项的问题，不是代码的问题。更改了编译选项就好了，
给点建议 能不能把Keil的编译选项直接拷贝过来啊，

github0null, 2019-12-11:
可以的，以后的版本会改进的

closed 2020-01-05 by github0null