github0null/eide #15: 编译配置还要重新配置,不会和原来keil工程一样.

fyyxxm, 2019-12-24:
另外配置里的RAM和ROM的大小 也无法更改,    更改会单独成行.

fyyxxm, 2019-12-24:
还切还有个问题:

typedef union
{
    uint16_t rader_id;
    struct
    {
        uint8_t self_id;
        uint8_t line_id;  
    };       
}_RADER_ID;

这种会报错:
"d:\My ARM Project\H7_Radar\STM32H750_Radar\User\..\User\main.h", line 44: Error:  #3093: anonymous structs are only supported in --gnu mode, or when enabled with #pragma anon_unions
      };

因为没有用到GUN扩展,配置里也没有这个选项.
KEIL里在C/C++配置里有 GUN extensions,而且还是默认勾选的.

fyyxxm, 2019-12-24:
#pragma anon_unions  keil中加这有效,   在VSCODE EDIE里加了无效````

fyyxxm, 2019-12-24:
#pragma anon_unions ,
keil里不选GUN extensions,然后加这有效, 在VSCODE EDIE里加了无效````

fyyxxm, 2019-12-24:
还有一个问题,也是编译器问题:
#ifndef __TARGET_FPU_VFP,
	#error This port can only be used when the project options are configured to enable hardware floating point support.
#endif

这段代码是freertos的port.c里的,
keil虽然显示错误,但是编译却不会报错,因为编译的时候自动添加:定义了__TARGET_FPU_VFP,   
但是在EIDE里却会报错,提示__TARGET_FPU_VFP未定义.

这类keil编译时自动添加的定义类型,还有很多.

fyyxxm, 2019-12-24:
#pragma anon_unions ,
keil里不选GUN extensions,然后加这有效, 在VSCODE EDIE里直接加无效````
但是在keil中加上后,,重新导入到EIDE里,EDIE就有效了,
说明在VSCODE EDIE里的更改没有效果```(实际也是:提示44行错误,我把代码移动到80行,还是提示44行错误,也更新依赖项了,代码位于main.h里)

github0null, 2019-12-24:
keil 版本太多，不同版本的项目文件各不相同，某些字段没法知道对应的是哪个配置，所以没有直接导入所有的 Keil 项目配置，需要自己配置，（也就几项而已）

1. 如果为项目安装了 Keil 包，ROM 和  RAM 会使用 KEIL 包里的ROM和RAM 地址，更改已经没有必要了，如果硬要更改必须卸载 Keil 包或者重新选择 芯片类型；

2. 后续会进行添加

3. 要添加自定义的 `宏` ， 点击 项目依赖->自定义依赖->默认->宏列表 进行添加

4. 暂时不清楚什么原因，（猜测 是因为 main.h 所在的路径有两个，一个是原来 Keil 项目下的，第二个是 EIDE 项目下的，但在编译时选择了前者，所以更改 EIDE 项目下的 main.h 是无效的）等找到了原因会进行修复的，`当然可以到直接 项目依赖->自定义依赖->默认->头文件列表 删除掉 旧项目的头文件路径`

感谢你的反馈，其他问题会逐步改进的

fyyxxm, 2019-12-24:
1.没有安装keil包, 默认的是M3 stm32f103c8, 如果使用其他的M3,M4或者M7,RAM和ROM就不对了.
2.
3.这种宏,keil 加不加都能通过编译,但是EIDE,不加报错,加了会报更多的错.这种宏比较特殊,和内核绑定的
4.

fyyxxm, 2019-12-24:
pack包,安装, 也没有任何现象,也不知道有没有安装成功.和没安装之前没任何区别````
应该是没安装成功,keil 5

另外错误追踪,也有个问题:
目录不能有空格:
例如:c:/my arm project/test这样的,
有空格就会错误追踪无效

这个工具很好,如果成熟的话,应该很受欢迎,
毕竟keil的编辑能力和VSCode不是一个级别的`````

github0null, 2019-12-24:
这些问题有些奇怪，如果安装pack成功了，可以展开 package 项，看到包的信息，接下来可以安装包自带的 依赖项，即使安装失败了是可以看到提示的，也许是插件发生了异常， 在项目的.EIDE/log目录下有一个日志文件：cl.eide.log 可以发上来看看

另外如果方便的话可以把失败的 Keil 工程打包发上来，帮助我来进行分析，不方便的话就算了

> 有些时候由于window的权限问题，没办法访问 c: 盘下的目录和文件，所以项目的位置最好要在其他盘里

fyyxxm, 2019-12-24:
https://github.com/fyyxxm/VScode_test

keil文件,把主要的东西删了.不影响.你试下把.

fyyxxm, 2019-12-24:
[cl.eide] : [4C4C4544-0034-5010-804B-B6C04F335032] : log at : 2019/12/24|20:08:18|GMT+0800

[cl.eide] : [4C4C4544-0034-5010-804B-B6C04F335032] : log at : 2019/12/24|20:08:24|GMT+0800

[cl.eide] : [4C4C4544-0034-5010-804B-B6C04F335032] : log at : 2019/12/24|20:12:50|GMT+0800

上面的就是log信息.

fyyxxm, 2019-12-24:
[cl.eide] : [4C4C4544-0034-5010-804B-B6C04F335032] : log at : 2019/12/24|20:08:18|GMT+0800

[cl.eide] : [4C4C4544-0034-5010-804B-B6C04F335032] : log at : 2019/12/24|20:08:24|GMT+0800

[cl.eide] : [4C4C4544-0034-5010-804B-B6C04F335032] : log at : 2019/12/24|20:12:50|GMT+0800

上面的就是log信息.
加载pack包,只会在VScode里 资源管理器(文本工作区)那里显示,但是也没法配置RAM和ROM呀.
EIDE里面的包是没任何东西的

github0null, 2019-12-24:
下图是安装完后的 package 视图
[image]

如果安装完 Keil 包，ROM 和 RAM 就 会使用 Keil 包里提供的，就不用再自己配置了，如果点击更改RAM和ROM，会出现以下提示
[image]

fyyxxm, 2019-12-24:
实测：只能装F1的pack包！！！！！！F4 H7都不能！！！！！！！！！试了很多次

github0null, 2019-12-24:
好吧，我还没试过，我以为这种包格式都应该差不多，看来又有区别，那可能是插件问题了，看来又得修bug了😭

fyyxxm, 2019-12-25:
总结下:
问题:
1.pack包只支持F1.
2.项目文件目录有空格时,错误追踪无效
3.非F1时,更改文件内容时,实际无效.需要把文件依赖的文件目录删掉,重新加载依赖项才行.
4.编译配置无GUN extensions选项
5.一些特殊宏定义会报错:如 __TARGET_FPU_VFP
6.debug时,出现svd文件解析错误提示,无法解决

建议:
1.加载PACK包后,选择相应的芯片后,  编译配置:会同步该芯片进行更新, 但是烧录选项信息并没有同步芯片.
  建议烧录选项也可以同步芯片信息,并且在编译成功后,自动加载hex烧录文件

2.编译选项:  建议增加keil里的bulid(只编译修改过的文件),否则有些大文件,全部编译很浪费时间.

3.下载烧录,目前下载程序不能reset and run,建议增加该选项.   现在launchAPPAfterProgram,选择ture 也无效.

4.虽然我用debug时失败,但是还是有个建议,建议debug时,debug的配置也能和自己选的芯片一致

github0null, 2019-12-25:
感谢你的建议
关于 package 包的问题已经解决，是一个小bug，其他问题正在解决中...
debugger 的 svd 如果一直失败可以不加， 主要是查看外设寄存器的，当然这个问题以后也会解决的
****
2.关于增量编译的问题，我暂时还没有研究清楚 Keil 是怎么做的，待我搞清楚了会加上的
3.launchAppAfterProgram 选项是我在 Jflash 脚本中看见的选项，顺手加上去的，至于是否真的复位了暂不清楚，当然这也很好解决
4. 由于 EIDE 和 stm32-debugger 是分离的两个插件，它们之间并没有什么关联，它们的配置相同点只有一项 `芯片类型`，而且两个不同的插件的配置纠缠不清我觉得也不是太好，容易引出新的bug

fyyxxm, 2019-12-26:
今天无视SVD错误试了下
debug的问题:

问题:
1.不提前加断点就仿真,暂停不了,只能 停止
2.提前加了断点,会提示:无法加载源“e:	est\STM32F103_Mode\VS\LL_NEW\src\USER\main.c”。
   并且使用 单步跳过,单步调试等功能时,并没有仿真到哪一行的指示.  完全不知道仿真到那里了.(根据板子现象,仿真是有效的)
3.需要仿真时,观察变量值的变化,添加到:求值也没用,只能添加到:监视.  并且再监视里也显示不可用,  实际变量值也仿真不出来````

fyyxxm, 2019-12-26:
芯片是:STM32F105

fyyxxm, 2019-12-26:
其实main.c的目录是  “e:\test\STM32F103_Mode\VS\LL_NEW\src\USER\main.c”,把没识别出来\t

github0null, 2019-12-26:
调试器不会在入口处停止，不像 keil 那样，

 对于stm32f10x的工程，我调试过很多,路径中有空格什么的，都没发现什么问题，除了不能在运行时暂停和设置条件断点之外，其他功能都有效，

必须要保证提供的.axf调试文件和.c文件都是最新的，即使.c和.axf文件的路径发生了变化，.axf也要重新生成，否则源是无效的，加断点也没用

调试器是GDB的，调试服务器JLinkGDBServer

fyyxxm, 2019-12-26:
1.不加断点启动debug调试,就不能停止,这肯定不合理的.

2.的确是识别不了\t.
我把test换成mytest, 就可以了

fyyxxm, 2019-12-26:
不是停止,是暂停

github0null, 2020-01-02:
你好，你上传的测试工程暂时没什么问题了，等其他细节完善了，测试完成了，就可以更新了
***
[image]

github0null, 2020-01-02:
> 对于 __TARGET_FPU_VFP 问题
其实是编译时我忘了为 Cortex-M4 以及 M7 系列的加上硬件浮点选项,  加上就没问题了

closed 2020-01-14 by github0null