github0null/eide #16: 1.07 问题!!!!

fyyxxm, 2020-01-04:
新导入EIDE,不会生成依赖项所需的路径和文件!!!!!!!!!!!!

你新建个试下就明白了.
用1.06的工程没问题
[image]

fyyxxm, 2020-01-04:
试了两台电脑,
F1和F7都试了,结果一样.
就是没生成依赖项 所需要的: 目录和文件.

1.06的工程都是包含的.

fyyxxm, 2020-01-04:
目前就发现两个问题,
1.
   1.07不会自动生成dependence文件夹.到此依赖项报错(手动添加dependence文件夹后,正常)
2.
   设置编译配置里的languageMode时,有时候 会报错(1.06也有此现象). 具体是 提示设置的语言模式后面有乱码,如两面两图, 感觉是语言模式保存时,存储的有问题,导致读取的不对.
[image]
[image]

fyyxxm, 2020-01-04:
上述问题2的,貌似旨在H7会有,F1没有. H7在设置languagemode之后,再设置 硬件浮点,就会有这种乱码错误.

还有 又发现个问题:
问题3:
 H7  不能下载,下载有错误!!!!!   提示如下!!!     F1正常!!! F4没测

[ INFO ] : ---------start---------- 2020/1/4-16:30:10-GMT+0800 ---------------------
[ TXT ] : Application log started
[ TXT ] :  - J-Flash V6.50 (J-Flash compiled Aug 16 2019 14:32:31)
[ TXT ] :  - JLinkARM.dll V6.50 (DLL compiled Aug 16 2019 14:30:54)
[ TXT ] : Opening project file [e:\VScodeProject\VSCode\Radar_Ranging\out\Radar_Ranging.jflash] ...
[ SUCCESS ] :  - Project opened successfully
[ TXT ] : Opening data file [e:\VScodeProject\VSCode\Radar_Ranging\out\Radar_Ranging.hex] ...
[ SUCCESS ] :  - Data file opened successfully (30412 bytes, 2 ranges, CRC of data = 0x450A284B, CRC of file = 0xF0D7EA9B)
[ TXT ] : Auto programming target (30412 bytes, 2 ranges) ...
[ TXT ] :  - Connecting ...
[ ERROR ] :     - ERROR: Could not detect CPU clock speed. Download of test code into RAM failed.
[ ERROR ] :     - ERROR: Failed to measure CPU clock frequency
[ ERROR ] :     - ERROR: Could not start CPU core. (ErrorCode: -1)
[ ERROR ] :     - ERROR: Failed to connect.
[ TXT ] : Could not determine CPU clock frequency.
[ ERROR ] :  - ERROR: Could not auto program target, not connected
[ ERROR ] : Error: Could not detect CPU clock speed. Download of test code into RAM failed.
[ TXT ] : Failed to measure CPU clock frequency
[ ERROR ] : Could not start CPU core. (ErrorCode: -1)
[ TXT ] : Failed to connect.
[ TXT ] : Could not determine CPU clock frequency.
[ TXT ] : Could not auto program target, not connected
[ TXT ] : Close project
[ TXT ] :  - Project closed
[ TXT ] : exit with code: 1
[ INFO ] : ----------------- exit ---- Upload failed --------------------------

github0null, 2020-01-04:
好的, 已收到，上述问题会解决的
> 只是下载问题暂不清楚，没用过H7的板子，可以使用备用的Jlink下载工具 下载选项-> useGUIDownloader 试试

fyyxxm, 2020-01-04:
H7 或者其他型号,用jlink的 j-flash下载没任何问题的.  应该还是有些参数的问题.

github0null, 2020-01-04:
> 下载问题暂时还没搞清楚（没板子不知道到底啥毛病， 看错误提示也没用）。
> 目前先把严重的问题更新了，下载问题以后再来处理，先用备用的将就一下😀

github0null, 2020-01-04:
> 此外 stm32-debugger 的bug也更新完毕，
> 其中 svd文件解析失败并不会影响到调试的运行，因为有的svd格式不太规范，总会出问题。svd 文件只是用来查看外设寄存器的值，而且这会影响到调试器的性能，因为要进行大量的读寄存器操作

fyyxxm, 2020-01-06:
现在debug虽然开始可以停止在开头,但是全速运行后,还是不可以暂停呀.  
商店里的Cortex-Debug是都可以的,也是jlink

github0null, 2020-01-06:
我禁用了 暂停 功能，以后再加，
> 感觉很少会用到暂停功能，以后会加上的

fyyxxm, 2020-01-06:
当路径有空格,  错误追踪还是无效!!!!! 没有空格时,是正常的,是可以追踪的!
[image]

github0null, 2020-01-06:
> 暂不清楚这是什么问题。

fyyxxm, 2020-01-25:
中文路径问题，还是没有完全解决：比如KEIL项目路径文件夹名字为：STM32F103开发工程模板，然后用EIDE导入后，编译会出现XX.h 找不到的问题

github0null, 2020-01-25:
在 v1.1.1 里我也发现了，原因好像是搜索路径少了些，你可以打开 头文件列表看看 ，如果缺了可以自己手动添加进去，

新版本我没发布, 自己在用顺便找找bug，老是在更新不太好,  等放假结束了再发

这个应该和中文没关系，我的项目路径也都是中文的，Keil 的编译器是支持中文路径的，而且我并没有任何编码处理

closed 2020-02-10 by github0null