github0null/eide #17: 无法编译C51源代码，路径错误

Kashouryo, 2020-01-05:
这个是我遇到的问题，项目是一个新项目，没有做任何改动

除此之外，EIDE好像并不会创建startup.a51

[ INFO ] Source file list:

 WARNING  Not found C51 Startup File! File must be named like this: xxx.A51

> C File:  c:\GitHub\8051fun\p00blink\.\src\main.c

[ INFO ] --------------------Start Compile...--------------------

C51 COMPILER V9.59.0.0 - SN: K1FMC-BW3IIC
COPYRIGHT Copyright (C) 2012 - 2018 ARM Ltd and ARM Germany GmbH. All rights reserved.

C51 FATAL-ERROR -
  ACTION:  PARSING INVOKE-/#PRAGMA-LINE
  LINE:    c:\Keil_v5\C51\BIN\C51.exe c:\GitHub\8051fun\p00blink\.\src\main.c SMALL OMF2 ROM(LARGE) OPTIMIZE(8,SPEED) REGFILE(c:\GitHub\8051fun\p00blink\.\out\main.ORC) INCDIR(c:\Keil_v5\C51\INC;c:\GitHub\8051fun\p00blink\.\dependence\build-in\STD;c:\GitHub\8051fun\p00blink\.\dependence\build-in;c:\GitHub\8051fun\p00blink\.\dependence) DEFINE(
  ERROR:   NON-NULL ARGUMENT EXPECTED
C51 TERMINATED.

 ERROR  生成失败(build Failed) : Compile C File:  c:\GitHub\8051fun\p00blink\.\src\main.c Error

github0null, 2020-01-05:
打开 自定义依赖-> 宏列表 任意添加一个宏进行编译就行了，这是构建工具的问题，会修复的

github0null, 2020-01-05:
startup.a51 可以不要，keil 的 LIB 目录下有很多 .a51 文件，自己复制一个过来就行了

Kashouryo, 2020-01-05:
非常感谢！现在已经可以成功编译。上面提及到会修复这个问题，我就先close了。

Kashouryo, 2020-01-05:
@github0null 我还有一个小问题，我看到有一些exe（例如说C51_Builder.exe）并非MIT license开源。请问这些可执行文件的license是什么。

closed 2020-01-05 by Kashouryo

github0null, 2020-01-05:
所有子目录下的文件都使用根目录下的 license

>  C51_Builder.exe 和 ARM_Builder.exe 是我用 C++ 编写的构建工具，使用他们需要特殊的命令, 通常这些命令由 eide 自动生成, 在编译时你可以看到 powershell 产生的调用命令

Kashouryo, 2020-01-06:
其实这里连js源代码都没有，无论如何都不算MIT吧...

github0null, 2020-01-06:
发布插件是必须要使用储存库的，因此这里只是一个临时的储存位置，用于存放，readme.md 和其他资源的，很多其他的插件都这样做

等待eide变得稳定起来，再开放代码也不迟，

Kashouryo, 2020-08-03:
如果开源了linux port也不难了