github0null/eide #19: 编译失败，找不到.A51文件

wherewhere, 2020-01-10:
小白一个，不知道发生了什么，别人用的都很正常，一脸懵逼。。。
```
[ INFO ] Source file list:
 WARNING  Not found C51 Startup File! File must be named like this: xxx.A51
 ERROR  生成失败(build Failed) : Not found source file
```

github0null, 2020-01-10:
> 看提示是没有找到源文件，可能是路径的权限问题，暂时不清楚
> 中文和带空格的路径是支持的，我的就没问题

[image]

wherewhere, 2020-01-10:
> > 看提示是没有找到源文件，可能是路径的权限问题，暂时不清楚
> > 中文和带空格的路径是支持的，我的就没问题
> 
> [image]

是不是keil在C盘的原因呢？

github0null, 2020-01-10:
有可能是的，c 盘是系统目录，在访问其他目录时有时候没有权限访问，就会导致未知的问题
建议把 工程，Keil 安装在 其他盘

github0null, 2020-01-10:
工程直接移动一下文件夹就行

wherewhere, 2020-01-10:
> 有可能是的，c 盘是系统目录，在访问其他目录时有时候没有权限访问，就会导致未知的问题
> 建议把 工程，Keil 安装在 其他盘

表示我就一个盘😂😂😂

github0null, 2020-01-10:
在C盘根目录自己新建文件夹也是可以的，自己新建的默认没有权限要求

wherewhere, 2020-01-10:
> 在C盘根目录自己新建文件夹也是可以的，自己新建的默认没有权限要求

好的，我试试。。。

wherewhere, 2020-01-10:
放在SD卡里也不行。。。。
[image]

github0null, 2020-01-10:
这就很奇怪了，还没遇到过这样的问题。。。

wherewhere, 2020-01-12:
> 这就很奇怪了，还没遇到过这样的问题。。。

很奇怪，第一次编译可以成功，第二次开始就找不到文件了。。。
[image]

github0null, 2020-01-13:
这个问题暂时不清楚
> 因为这个问题没法在我的电脑上重现，没法知道问题在哪

github0null, 2020-01-14:
建议换台电脑试试

closed 2020-01-14 by github0null