github0null/eide #21: 编译C51失败，如下图

kklili, 2020-01-13:
配置完成后，编译C51，提示找不到 reg52.h，错误信息如下，请问是怎么回事？

> C File:  f:\C51\chaoshengbo\目标 1\.\src\源组 1\main.c

[ INFO ] --------------------Start Compile...--------------------

C51 COMPILER V9.56.0.0 - SN: K1FMC-G68IKC
COPYRIGHT Copyright (C) 2012 - 2016 ARM Ltd and ARM Germany GmbH. All rights reserved.
*** WARNING C318 IN LINE 1 OF f:\C51\chaoshengbo\目标 1\.\src\源组 1\main.c: can't open file 'reg52.h'
*** ERROR C202 IN LINE 2 OF f:\C51\chaoshengbo\目标 1\.\src\源组 1\main.c: 'P1': undefined identifier
*** ERROR C202 IN LINE 17 OF f:\C51\chaoshengbo\目标 1\.\src\源组 1\main.c: 'LED0': undefined identifier
*** ERROR C202 IN LINE 19 OF f:\C51\chaoshengbo\目标 1\.\src\源组 1\main.c: 'LED0': undefined identifier

C51 COMPILATION COMPLETE.  1 WARNING(S),  3 ERROR(S)

 ERROR  生成失败(build Failed) : Compile C File:  f:\C51\chaoshengbo\目标 1\.\src\源组 1\main.c Error

PS F:\C51\chaoshengbo\目标 1>

github0null, 2020-01-13:
reg52.h 由 Keil 提供，一般在 `[keil 安装目录]\C51\INC` 下找到 REG52.H， EIDE 会自动使用该目录作为内置的头文件目录，如果没找到， 可能是 Keil C51 的问题

> 点开 内置依赖->头文件列表 会有 内置的头文件目录 的位置

[image]

kklili, 2020-01-13:
你好，感谢回复。我看了一下，我这边头文件目录是空的，我的keil都是正常安装的，也能正常使用的。安装路径是 G:\Keil_v5\C51 ，配置的时候选择的也是G:\Keil_v5\C51\bin下的tools.ini。 请教一下，这个头文件目录能手动设置吗？

github0null, 2020-01-13:
**1.** 可能是 TOOLS.INI 选错了，在 TOOLS.INI 文件所在的目录下，会有一个 C51的文件夹，如下图
[image]

**2.** 头文件目录可以在 自定义依赖->头文件列表中添加

kklili, 2020-01-13:
您好，我尝试添加keil目录下的tools.ini，但是编译的时候报错说C51设置路径不正确。我用C51\bin目录下的tools.ini,手动添加了C51下的inc目录，可以编译成功，但是生成hex的时候出错哦，我看了一下，是用lib文件的目录有错。
[image]
[image]

github0null, 2020-01-13:
看起来不支持你这个版本的 Keil , 因为你这个版本的 Keil 安装目录的结构就和我的不同，肯定会出问题的，我的 keil 安装目录下没有 BIN 这个文件夹

github0null, 2020-01-13:
下图是我安装的版本
[image]

kklili, 2020-01-13:
我看了一下我的目录结构，和你的差不多啊。 
[image]
[image]
[image]

下面是keil根目录下的tools.ini文件和c51/bin目录下的tools.ini文件，您看看有没有参考价值。

[tools.zip](https://github.com/github0null/eide/files/4053675/tools.zip)

github0null, 2020-01-13:
你的TOOLS.INI位置不对。

你可以把bin目录下的TOOLS.INI 移动到 G:\keil_v5，移动的时候改一下名字，随便起，后缀别改就行，然后再eide中重新设置一下

github0null, 2020-01-13:
这个问题，后续我会改进的

kklili, 2020-01-13:
OK 照你说的，移动tools.ini后 目前就好了。 感谢啊

kklili, 2020-01-13:
另外，我有时候导入keil的工程，生成的工程文件这样的，然后在eide里面打开就没有反应。

[image]
用文本打开工程文件的内容如下：
[image]

{"folders":[{"path":"."}],"settings":{}}

github0null, 2020-01-14:
工程文件是.\.EIDE\EIDE.json，你发的是工作区文件，发一下截图

打开工程失败，没有什么提示吗，打开失败一般会有一个警告提示，

如果没有提示可以打开.\.EIDE\log\eide.log, 看看日志中有没有记录什么错误

kklili, 2020-01-14:
选择打开那个项目文件，没有任何反应和提示，我看了一下，也没有eide.log文件。我把整个目录上次您看一下吧。
[19-1、LED闪烁.zip](https://github.com/github0null/eide/files/4057158/19-1.LED.zip)
下面是项目的路径
[image]

github0null, 2020-01-14:
这可能是上次更新后的一个小bug，你需要先关闭 AutoGenerateRTE_Components选项, 插件设置里有

kklili, 2020-01-14:
OK 关闭之后就正常了，多谢了！

closed 2020-01-14 by github0null