github0null/eide #24: EIDE基本成熟了。NICE！！！

fyyxxm, 2020-02-03:
NICE! 1.13
现在个人认为可以考虑：
1.下载后可以自动复位的功能
2.下载路径，编译成功后自动加载
3.可选ARM Compiler v6编译器，现在用V6也很多了  （V6的配置和V5不同，所配置的东西也不一样）
4.自动生成Cortex-Debug调试配置，可选择Jlink openocd stlink（stutil）等其他的

还有log可以换鲜明一些的，感觉比较正式。

现在还有假期，没事的时候，可以研究研究。。。。。

github0null, 2020-02-03:
4.Cortex-Debug 的其他选项手动配置会方便些, 插件只会帮你设置好elf和svd的路径、芯片类型、cpu类型，其他选项不管；如果没有发现 cortex-debug 配置项，则会创建一个默认的，默认的是 jlink

其他功能会慢慢实现的

感谢支持

fyyxxm, 2020-02-10:
1.14 导入keil工程，根本导入成功不了。
但是的确会在保存的文件夹内，新建一些文件。

github0null, 2020-02-10:
已经发了更新

因为修改了许多东西，所以有点小 bug

fyyxxm, 2020-02-10:
刚才测了下.
1. 实测 找不到.h文件。 试了F1 F4 H7，路径有无中文，带不带RTOS。都报错，说.h找不到···不同的.h头文件，试了几次。例如带RTOS时，找不到FREERTOS.H， 我在依赖里，把包含这个头文件的路径删掉，重新加载，还是不行。
2.（非重点），如果keil里的文件是UTF-8格式的，包含中文，通过EIDE导入VScode里，会出现乱码。通过VSCODE设置文件格式为UTF-8或者GBK 也依然是乱码····

不要觉得我烦哈，我只是帮你测试下，提交问题。。。。。

github0null, 2020-02-10:
1. 这个问题不够详细，我导入了 RT-Thread 里的模板工程 atk-explorer 导入完成直接编译和 keil 直接编译的效果一样，
我自己的一些工程在发布前也试过了，没找到问题出在哪里，

2.这个情况在之前的更新里提到过，无BOM头的 UTF-8 很难识别，所以就算是 utf8 的也会再转换一遍，但是大部分keil工程默认都是GBK的，所以一般都是正常的，这个问题以后会解决的

很高兴你能提出意见，这会帮助我把这个工具做的更好（反正在家也很无聊），主要是我自己的工程太少太简陋了，有很多细节问题没办法在我的工程里重现

fyyxxm, 2020-02-11:
问题就是编译的时候找不到.h头文件。
给你几个模板，上次的H7的你也可以继续验证，测试的时候可以都试试，编译。  直接克隆都可以到本地了。
https://e.coding.net/fyyxxm-01/STM32F429_Mode.git
https://e.coding.net/fyyxxm-01/STM32F429_Mode_FreeRTOS.git
https://e.coding.net/fyyxxm-01/STM32F103_Mode.git

fyyxxm, 2020-02-11:
找不到头文件的问题，是因为我用错编译器了。。。
现在编译就是这，该怎么设置呢。吧AXF转ELF关掉了，还是这
E:\git\first_case\f1\22\LL_NEW>& 'c:\Users\fyyxx\.vscode\extensions\cl.eide-1.2.0\bin\ARM\ARM_Builder.exe' -b 'D:\Keil_v5' 'e:\git\first_case\f1\22\LL_NEW\out\LL_NEW.params'
& was unexpected at this time.

E:\git\first_case\f1\22\LL_NEW>& 'D:\Keil_v5\ARM\ARMCC\bin\fromelf.exe' --bin --output 'e:\git\first_case\f1\22\LL_NEW\out\LL_NEW.bin' 'e:\git\first_case\f1\22\LL_NEW\out\LL_NEW.axf'
& was unexpected at this time.

github0null, 2020-02-11:
终端是Powershell

fyyxxm, 2020-02-11:
两个小问题：
项目依赖里的宏定义：最好默认加个__CC_ARM （仅限AC5，AC6不是这个宏定义）
因为加了这个，编辑文件的时候，就可以自动联想补全代码了，不加的话，都会被认为 未定义，所以很多系统函数不能自动联想和补全。  加不加，都不影响编译和结果。

还有个，编译选项的RAM和ROM layout，自动生成的和KEIL里的不一样，里面的值是一样的，但是默认勾选是不一样的！！！！虽然不影响编译，但是影响程序执行。 （STM32H743II实测）

上面两个问题，可改可不改，毕竟可以自己动手改，不改的话最好在教程里 说明下。

github0null, 2020-02-11:
1. 可以在插件设置里自己加, release/Debug 的宏列表(设置有全局和仅工作区的)，设置好之后切换一下模式或者重开一下项目就能完成添加

2. 看了以下，你说的应该是 启动ROM 没有正确勾选, 这是包描述文件参数导致的，
[image]
上图 startup 字段 标记了这段ROM是不是默认的启动ROM，两个都选中了，但是启动ROM只允许有一个，所以插件里后面的选项覆盖了前面的，不知道 Keil 是怎么判断的，**下次更新如果两个都选直接默认第一个就行了**

github0null, 2020-02-11:
补充一下，导入时的编码问题，原来设了一个选项 AutoConvertFile， 你可以在导入之前关掉它，这样在导入时不会作任何编码转换，你可以试一下

fyyxxm, 2020-02-11:
是的，但是RAM其实也错了，H743II EIDE默认里默认吧off chip里的RAM也全部勾选了，KEIL只是把on chip里的RAM勾选。
我说的RAM勾选是前面的default勾选，非后面的nolinit勾选

github0null, 2020-02-11:
确实是的，刚才没注意看RAM，RAM 和 ROM 一样描述文件里都默认选中了，我再改一下就行了

[image]

closed 2020-02-22 by github0null