github0null/eide #27: 遇到一个比较影响使用的问题，不知道是不是个例

ethan-www, 2020-04-13:
- 如果项目中有比较长的文件名，那么右侧的一些功能按钮都会被顶到最右边被主窗口覆盖而没办法使用。
正常有功能按钮的窗口：
[image]
只要展开的内容中有超过窗口长度的右侧按钮就会被遮住：
[image]
展开一个长路径后右侧按钮全都被顶没了，即使窗口拉到最最大右侧按钮也全部出不来
[image]

github0null, 2020-04-13:
感谢你的反馈：

1.关于路径问题，我将会用相对路径替换，以减小长度；2.项目名称过长导致的遮挡，无法避免；解决方法：关闭项目，将项目的根目录重命名为较短名称后重新打开项目即可；

这种问题本不属于插件的 bug，而是你的路径问题，应采用简洁明确的路径储存项目

ethan-www, 2020-04-15:
> 感谢你的反馈： 1.关于路径问题，我将会用相对路径替换，以减小长度；2.项目名称过长导致的遮挡，无法避免；解决方法：关闭项目，将项目的根目录重命名为较短名称后重新打开项目即可； 这种问题本不属于插件的 bug，而是你的路径问题，应采用简洁明确的路径储存项目

感谢大神提供了这么好用的插件，
1.10版烧录选项那的长路径己经OK了，头文件列表那还是用的绝对路径，会有遮挡。
项目根目录重命名这个好处理，要全面的缩短整个路径长度和层次就有些不好弄了

另外从keil导入的项目，目前是复制.c文件到新目录，但是直接引用的原项目.h目录。这样子我导入后再改代码时就会出现.c改动保存的文件在复制的新目录里，.h改动保存的文件在原项目目录里。
这样子感觉有些乱，不利于文件管理，是否可以改成除keil目录外项目目录下的.c和.h都复制新目录下，或者都不复制直接引用原项目目录下的文件呢？

github0null, 2020-04-15:
[image]
***
头文件目录，有些不能使用绝对路径，因此即使显示成相对的，也只是一部分，这样做并不能完全解决这类问题，其实只要鼠标悬停在项上，那些按钮就会出现，也会有 tooltips 供你查看路径，视图的显示是由 vscode 控制的。
***
keil 项目没有给足够的信息来帮助我复制头文件，keil 也只是简单的保存了一下头文件目录的路径而已，频繁更改的头文件应该由你自己复制过去，并去掉插件头文件路径中的相关路径

导入问题其他人也说过这类问题，由于 keil 和 此插件的需要的项目结构不同，暂时不能直接引用 keil 项目的信息。

ethan-www, 2020-04-15:
[image]
主要是我这里如果路径超长，鼠标悬停在项上时，那些按钮并不会出现，不知道是不是哪里设置问题。
-------------------------------------------------------------------------------
那么头文件复制是否可以直接复制除路径中带keil关键字之外的所有include目录，
或者在头文件目录列表上做一个复制安装到此插件项目路径下的按钮呢？

ethan-www, 2020-04-15:
[image]
不显示按钮的问题找到了，在设置中去掉：workbench.list.horizontalScrolling 前面的钩钩，鼠标悬停时按钮就能出来了，
谢谢！

github0null, 2020-04-15:
自动复制头文件，没法知道到底要复制到什么样的目录结构中去，
之所以源文件能够帮你复制到较合适的路径，是因为 keil 为源文件设置了组信息；
等我有了较好的办法再来解决这个问题；
目前来说，手动复制添加更加灵活稳定；

closed 2020-04-19 by github0null