github0null/eide #28: 按照教程导入了一个工程，却无法编译，Init params failed，提示未能找到路径

tangs92, 2020-04-19:
按照教程导入KEIL项目

keil版本 529
编译的芯片 ST的F765ZIT6

安装完F7的pack包(Keil.STM32F7xx_DFP.2.12.0.pack)
未修改其他任何的设置
无法编译，提示如下错误

PS xxxxxx[以上为项目路径，省略]> & c:\Users\Administrator\.vscode\extensions\cl.eide-1.10.1\bin\ARM\General_Builder_V6.exe -b D:\Keil_v5\ARM -d e:\xxxxxx[以上为项目路径，省略]\.EIDE -M c:\Users\Administrator\.vscode\extensions\cl.eide-1.10.1\bin\ARM\arm.v5.model.json -p e:xxxxxx[以上为项目路径，省略]\out\arm_builder.params -o e:xxxxxx[以上为项目路径，省略]\out -m Normal-MULTHREAD
 ERROR  Init params failed !, 未能找到路径“xxxxxx[以上为项目路径，省略]\out\arm_builder.params”的一部分。

github0null, 2020-04-19:
很明显这是路径问题，是 C# 没找到路径抛出的异常；
虽然路径是支持空格和中文的，但最好使用全英文的路径；
发一张命令行的截图上来看看，

tangs92, 2020-04-25:
我换了英文路径后，它可以编译了，谢谢

closed 2020-04-25 by tangs92