github0null/eide #35: 打开串口监控时建议增加波特率设置

heartacker, 2020-06-12:
你好。
EIDE的串口工具在点击打开时可以选择串口号。但是波特率都是固定的，不太灵活。
建议在选择打开串口号之后，增加一个串口波特率的设置，
可以设置为用户输入或者提供一个list，里面包含常用的波特率，如4800，9600,19200，115200,25600等，这样就比较方便了。
感谢

github0null, 2020-06-12:
串口的所有参数都是可以自定义的，打开工作区文件进行配置

heartacker, 2020-06-12:
> 串口的所有参数都是可以自定义的，打开工作区文件进行配置

明白这一点，但是经常还是需要改变不同的波特率进行测试的， 如果每次改波特率就去更改配置，略显累赘啊，
建议如下：
1. 用户可以配置默认波特率，
2. 选择串口之后，弹出波特率的选择的list。同时默认勾选用户的默认配置，
3. 用户可以上下键选择不同的波特率，直接回车即可（此选择不会更改默认分辨率）

从效果来看，打开串口，选择波特率，并不繁琐啊。要是实现了就很方便，希望考虑下哈哈
感谢

github0null, 2020-06-12:
很理解你的感受，你的要求十分简单实现；

但大多数人都很讨厌不断地弹出选项进行选择，他们都希望能够 **一键启动**
其实我正打算将 串口选择 也放到配置中；
***
其实通过修改 .code-workspace 文件中的配置要快得多，而且这样不同的工作区配置不会相互影响
并不是只能通过打开插件配置来修改

heartacker, 2020-06-13:
> 很理解你的感受，你的要求十分简单实现；
> 
> 但大多数人都很讨厌不断地弹出选项进行选择，他们都希望能够 **一键启动**
> 其实我正打算将 串口选择 也放到配置中；
> 
> 其实通过修改 .code-workspace 文件中的配置要快得多，而且这样不同的工作区配置不会相互影响
> 并不是只能通过打开插件配置来修改

明白你的用意!
如果你真的打算串口选择 也放到配置中；
那么建议加一个开关,这个开关默认`false`,
这个开关的作用:
- `false`: 直接按照配置的端口号和波特兰启动串口.
- `true`: 用户可以选择端口号,同时也在选择端口号之后增加我之前说的波特率设置.
👯

github0null, 2020-06-17:
这个以后会加上的

closed 2020-07-01 by github0null