github0null/eide #55: 使用问题

xbmo123, 2020-12-05:
1.新建工程，自己就在工作区打开了。之后，打开工程，EIDE_PROJECTS栏没任何显示。把工作区清空后，再打开工程，就可以显示在EIDE_PROJECTS栏了。

2. 按F7编译，总是报警warning: allocated section `RW_IRAM1' not in segment，打开调试运行不正确，程序跑飞。从模板下创建的工程也是一样。

3. 很想用这个插件，很多地方不懂怎么用，系统环境（win7?win10? 32bit? 64bit?）支持哪种？各种编译器都需要用到VSCODE中的哪些插件（必须的那种），各种编译器需要用到哪些要用户提供的文件，比如：.ld文件, .sct文件，makefile文件, 希望能有详细的使用说明。

github0null, 2020-12-05:
1. 打开项目的方法，【1】双击 vscode 工作区文件，【2】通过 "打开项目" 操作打开

2. 你所说的警告是 axf 转 elf 产生的，而不是编译过程的警告，这个可以忽略
    模板下的所有工程都经过我的测试没有问题，有啥问题你可以将 Cortex-debug 调试控制台信息截个图什么的发我看看

3. 支持 windows 代表支持 win7 及以上的任意版本
    关于其他插件，你只需要安装 C/C++ 和 Cortex-Debug 插件即可，如果有编译器你没下载，在用相应功能前会给出提示

**有很多问题已经在 使用手册 里面说明了，你可以去看看**
**最近我也在完善这个插件的文档，这个插件其实用起来也很简单的，没什么特别的操作**

xbmo123, 2020-12-06:
没想到大神这么神速。

[image]

[image]
[image]

xbmo123, 2020-12-06:
有个图发重复了

xbmo123, 2020-12-06:
上面的图是用VSCODE+EIDE+GNU Toolchain+CORTEX-DEBUG插件+JLINK套件来做测试。

github0null, 2020-12-06:
没问题的，下面的 gif 里你也能看到 PA 的 ODR1 在变化
用调试器暂停程序，看看程序到底跑到哪了，不能完全根据板子的现象来做判断

[image]

xbmo123, 2020-12-06:
多谢大神的耐心指导！
已验证了，编译没问题，问题在debug阶段。
我测试的芯片是上海海尔公司出的HR8P506。这个芯片不在JLINK支持的芯片列表内。我在烧录配置那里就选了一个各项指标和这个芯片差不多的来替代（在编译配置那里设置的ROM/RAM是对的），就是这一步导致了debug出错。
拿了个STM32的开发板来测了整个流程，新建-拉库-编APP-编译链接-下载-调试，的确整个环境没有任何问题。用着也很顺手。
我这么测试的目的是想用这一套工具能够通用于市场上所有的CORTEX-M内核的单片机，而不依赖于单片机厂家官方出的自己独有的工具，这样，我不用学多种编译器，不用学多种工具的使用，而又能够选择任何一款CORTEX-M内核单片机。
这款芯片官方出了一个KEIL用的插件，所以在KEIL下能够使用，当进入调试时JLINK检测到列表内没有这款芯片而让我重选芯片的界面，我就在这个界面内的下拉列表中选了个制造商=UNSPECIFIED  DEVICE=CORTEX-M0  CORE=CORTEX-M0的选项。结果也能用，希望EIDE也能支持这个选项（目前只有具体芯片型号的选项而没有仅指定内核的选项）。

github0null, 2020-12-06:
插件上的 调试 和 下载是不相干的，调试是 Cortex-debug 完成，在启动调试后，它有自己的下载程序，要正常调试你可以把 launch.json 的 device 字段填成 Cortex-M0 试试

eide 的下载工具只是单独用来下载的，和调试无关，jlink下载必须要知道芯片的名称，才知道芯片的 flash 起始地址和 大小，直接传递 Cortex-Mx  这种系列名作为芯片名称，命令行会出错的

closed 2020-12-26 by github0null