github0null/eide #57: EIDE Manual

github0null, 2020-12-06:
http://www.em-ide.com/docs/eide-manual/#/ 

 Description

github0null, 2020-12-06:
hello, everyone !

Yorkzero, 2020-12-08:
bookmark有点问题，无法快速定位文本

github0null, 2020-12-08:
> @Yorkzero
> bookmark有点问题，无法快速定位文本

哪个标签，我这很正常的

Yorkzero, 2020-12-09:
[image]
@github0null 
浏览器为chrome最新版本，有时能够跳转有时无效

alkimgokcen, 2020-12-14:
Hello, 

I am using a stm32FM33G048 processor. Is it possible to program on vsCode using EIDE extension.

github0null, 2020-12-14:
> @alkimgokcen
> Hello, 
> 
> I am using a stm32FM33G048 processor. Is it possible to program on vsCode using EIDE extension.

sure, it supports all cortex-m mcus

alkimgokcen, 2020-12-14:
> > @alkimgokcen
> > Hello,
> > I am using a stm32FM33G048 processor. Is it possible to program on vsCode using EIDE extension.
> 
> sure, it supports all cortex-m mcus

I cant find the stm32FM33G048 hadders, and cant find the mcu in the " CPU NAME " section.
And also, how can i create the template for related MCU

github0null, 2020-12-14:
> @alkimgokcen
> > > @alkimgokcen
> > > Hello,
> > > I am using a stm32FM33G048 processor. Is it possible to program on vsCode using EIDE extension.
> > 
> > sure, it supports all cortex-m mcus
> 
> I cant find the stm32FM33G048 hadders, and cant find the mcu in the " CPU NAME " section.
> And also, how can i create the template for related MCU

This plug-in requires you to organize the source files and structure yourself.

And all it does is compile your source code and generate the final axf/elf/hex/bin... files, and provide some interface to download your program.

so I didn't provide many templates because there were too many MCU

Yorkzero, 2020-12-16:
> > @Yorkzero
> > bookmark有点问题，无法快速定位文本
> 
> 哪个标签，我这很正常的

目前OK了，可以正常使用

closed 2021-02-13 by github0null