github0null/eide #58: 编译不成功

JinFaChen, 2020-12-07:
你好。在github上下载 stm8s103的demo工程编译，只有在 “TERMINAL”栏出现>Executing task in floder demo:"c:\user\PC2018\.vscode\extensions\cl.eide-1.19.0\bin\builder\Genneral_Builder_V6.exe -b"c:\Program Files\SDCC" -M sdcc.model.json -p c:project\demo_stm8s103_sdcc\demo\out\Debug\builder.params -m Fast-MULTHREAD"<,就没有其他提示信息了该如何解决

github0null, 2020-12-07:
构建工具没有启动吗 ?，是否安装了 .NET FrameWork 3.5 ?, 一般 win7 自带 .NET FrameWork 3.5

或者你将 Genneral_Builder_V6.exe 直接拖到 cmd 窗口里执行试试

JinFaChen, 2020-12-08:
> 构建工具没有启动吗 ?，是否安装了 .NET FrameWork 3.5 ?, 一般 win7 自带 .NET FrameWork 3.5
> 
> 或者你将 Genneral_Builder_V6.exe 直接拖到 cmd 窗口里执行试试

有.NET FrameWork 4.5.1。直接拖到cmd窗口显示 ERROR init build faild failed !some command line arguments missing System.Exception,在ARM_Builder_v6.program.main<string[] args> 
是还需要什么参数吗

github0null, 2020-12-08:
你点编译，Terminal 栏会输出命令行的，你直接复制到 cmd 里去执行就行
.NET 4 和 3 是不兼容的，最好装上 .NET 3.5，因为这程序是 .NET 3.5 的

closed 2021-01-19 by github0null