github0null/eide #60: 新建51工程，编译不成功

kl-xiao, 2020-12-15:
你好，我新建一个C51 SDCC编译器的工程，编译一直不成功，麻烦帮忙看下，谢谢啦

工程地址
https://github.com/kl-xiao/Print

一般报这个错误
ASlink-Warning-No definition of area HOME
ASlink-Warning-No definition of area XSEG
ASlink-Warning-No definition of area PSEG
?ASlink-Error-<cannot open> : ".\out\Debug\main.lst"
?ASlink-Error-<cannot open> : ".\out\Debug\gvm08_gpio.lst"
?ASlink-Error-<cannot open> : ".\out\Debug\gvm08_uart.lst"
?ASlink-Error-<cannot open> : ".\out\Debug\system_gvm08fx.lst"

有时候报错误这个（这个好像是因为每次重新打开的时候，之前头文件里面屏蔽掉的一些没用到的头文件恢复了）
rebuild时候报这个错误
>> compiling 'gvm08_gpio.c'
In file included from ./inc/gvm08_gpio.h:8,
                 from .\src\gvm08_gpio.c:1:
./inc/gvm08x.h:726:26: fatal error: gvm08_beep.h: No such file or directory
compilation terminated.

github0null, 2020-12-15:
1.  你没有添加包含路径，是找不到头文件的，在 项目依赖->包含目录 将 inc 目录加进来
2.  在 inc/gvm08x.h 里面注释掉没有在 inc 目录中发现的头文件

> 还有一点是 js 的 json 解析器是不支持注释的，不要在 json 中加注释，编译配置 afterbuildtask 里面可以直接用 disable 选项直接屏蔽

***
[image]

***
修改后的工程
[Print.zip](https://github.com/github0null/eide/files/5693051/Print.zip)

kl-xiao, 2020-12-15:
1.我包含了，可能上传的时候有问题
2.这个就是我问题里面说的注释掉之后，不知道为什么重新构建的时候，又没有被注释了

那个修改后的文件，我这边编译现在还是有问题，现在只报这个错误了，不知道什么原因
ASlink-Warning-No definition of area HOME
ASlink-Warning-No definition of area XSEG
ASlink-Warning-No definition of area PSEG

github0null, 2020-12-15:
你用的什么版本的 SDCC

github0null, 2020-12-15:
我这里没有这种问题

kl-xiao, 2020-12-15:
没看到版本号，更新日志里面最新是20160903的

kl-xiao, 2020-12-15:
3.6.3的，会有影响吗

github0null, 2020-12-15:
最好使用最新的，3.6.3 有点老了，可能会有 bug，而且过旧的版本不支持多文件编译
我使用的是 4.0
是2020 年发布的

kl-xiao, 2020-12-15:
好的，我试试，感谢啦

kl-xiao, 2020-12-15:
更新为4.0的SDCC编译器之后，能正常编译了，感谢大神的耐心解答！

closed 2020-12-15 by github0null