github0null/eide #64: EIDE Manual

github0null, 2021-02-13:
https://docs.em-ide.com/#/ 

 Description

github0null, 2021-02-15:
test

github0null, 2021-02-16:
test

github0null, 2021-04-02:
hello ?

wyh0106, 2021-04-02:
HELLO WORLD

zhang00956, 2021-04-07:
注意插件仅支持*.lib后缀库文件，如果工程中有*.o库文件，将后缀改成lib,否则编译完成链接不能通过。

zhang00956, 2021-04-12:
如果gcc编译成功，烧写进去程序死机，或者跑的不正常。在插件编译、清理位置的空白处右键点击（show compiler command）,会显示出编译命令、连接命令，对照着检查修改编译器选项中的配置。

zhang00956, 2021-04-12:
插件烧录配置如果没有自己用的芯片型号，要修改cl.eide-2.5.2\res\data\JLinkDevices.xml配置文件

heartacker, 2021-04-12:
你好,请问你这个文档系统是什么程序?gitbook吗?

github0null, 2021-04-13:
> @heartacker
> 你好,请问你这个文档系统是什么程序?gitbook吗?

docsify

apythonboy, 2021-04-20:
支持stm32mp157吗？Cortex-M4的

github0null, 2021-04-20:
@apythonboy 
> 支持stm32mp157吗？Cortex-M4的

支持的

apythonboy, 2021-04-23:
@github0null m4 + a7 双核的

yun-deram, 2021-05-07:
你好，新建89C52工程时，提示错误为表达式必须含有常量，编译没有问题，KEIL自带的.H文件也是有报同样的错误 sbit  LED = P0^0; P0下画红色波兰线
新建后必须在构建配置选择工具链路径，不然宝扩展错误
请问这是什么原因呢

yun-deram, 2021-05-07:
> @github0null
> > @yun-deram
> > 你好，新建89C52工程时，提示错误为表达式必须含有常量，编译没有问题，KEIL自带的.H文件也是有报同样的错误 sbit  LED = P0^0; P0下画红色波兰线
> > 新建后必须在构建配置选择工具链路径，不然宝扩展错误
> > 请问这是什么原因呢
> 
> c/c++ 扩展不能用在 c51 上，你应该屏蔽掉错误提示；
> 使用内置的模板创建工程，就不会这样
> 
> 

如果我在构建配置哪里选择了 Keil-C51 就不会报错，也没有红色波兰线，用默认的SDCC编译器就会报错，在新建空项目时没有默认新建 INC文件夹，

github0null, 2021-05-07:
> @yun-deram
> 
> > @github0null
> > > @yun-deram
> > > 你好，新建89C52工程时，提示错误为表达式必须含有常量，编译没有问题，KEIL自带的.H文件也是有报同样的错误 sbit  LED = P0^0; P0下画红色波兰线
> > > 新建后必须在构建配置选择工具链路径，不然宝扩展错误
> > > 请问这是什么原因呢
> > 
> > c/c++ 扩展不能用在 c51 上，你应该屏蔽掉错误提示；
> > 使用内置的模板创建工程，就不会这样
> > 
> > 
> 
> 如果我在构建配置哪里选择了 Keil-C51 就不会报错，也没有红色波兰线，用默认的SDCC编译器就会报错，在新建空项目时没有默认新建 INC文件夹，
> 

- KEIL_C51 的语法是带有扩展的 C，C/C++ 插件是不支持这种语法的，而 SDCC 是标准的，所以是可以的；
- 这种语法检查并不代表你的程序编译一定会失败，这些只是建议，跟 KEIL 不一样的
- 你在 *.code-workspace 文件的 settings 字段里加上 **"C_Cpp.errorSquiggles": "Disabled"** 即可屏蔽掉这种错误；

strbtn, 2021-06-11:
STC15W 可以调试吗  怎么配置?

Print2E, 2021-07-12:
Error : parse data error from 'SerialPortMonitor.exe'这个是什么问题导致的呢

dreamtale-whf, 2021-07-17:
你好，报了个case错误是不支持这种格式吗

".\mod\src\reg.c", line 379: Error:  #53: expected a ":"
                              case 1 ... 8 :
                                     ^

closed 2021-07-26 by github0null