github0null/eide #68: Can you provide English README by default?

GitMensch, 2021-07-26:
The reasoning is that this one is parsed by market places and is first shown to users in vscode, too.
Other extensions that use a native language either have their localized README in a link (as the current one has for English) or have a dual-language README.

github0null, 2021-07-26:
- The beginning of the README has a link to the English README

[image]

GitMensch, 2021-07-26:
Yes, I'm aware of this. The question was: could you swap that so the English one is the default and has the link to the Chinese one or adjust the README to include both languages?

Of course a "no" and closing would be ok, but using one of the options above would provide more visibility because the market places and search machines commonly index the README and more people speak English in general.

github0null, 2021-07-26:
Sorry !, I haven't written the full English document yet, just having the English README is not enough. So I didn't put English README first.

Rest assured, the extension market searches for plug-ins by tag, which I've provided in package.json

closed 2022-01-01 by github0null