github0null/eide #69: Please co-publish on Open VSIX Registry

GitMensch, 2021-07-26:
See https://github.com/eclipse/openvsx/wiki/Publishing-Extensions for some details. Currently no similar extension is found on open-vsx.org, which is also the default provider for VSCodium (which itself is the packaged vscode build in different GNU/Linux distros).
So far no extension for embedded development is available there and it looks like this extensions would be a nice addition.

github0null, 2021-07-26:
Okay, but what are the advantages of this thing?

[image]

GitMensch, 2021-07-26:
Thanks!

The advantage is that open-vsx can be used with any vscode based editor (and commonly also with theia as this provides most of the necessary api), while the ms marketplace may explicit only used with "Microsoft Visual Studio Code" (the binaries that are provided my MS and that does not match the vscode repo).
This is also the reason that those normally don't use the MS market place as the users would breach its license.

I can confirm that the extension is now usable in VSCodium. Thank you very much!

closed 2021-07-26 by GitMensch