github0null/eide #80: Add support to ARM SC000 core?

ckforce, 2021-12-15:
Can it be considered?

github0null, 2021-12-15:
Yes. But since i am not familiar with the other ARM family, I need some project examples so that I can understand the compilation parameters.

ckforce, 2021-12-16:
I can offer some example for you now, which from uVision 5.24.

### C compile parameter:
`-c --cpu SC000 -D__MICROLIB --li -g -O0 --apcs=interwork -I ../Src/Main/Inc -I ../Src/Lib/Inc -I ../Src/Drv/Inc -I ../Src/Arch -I ../Src/Alg/Inc -I ../Src/Arch/CPU -I ../Src/APP/Inc -I ../Src/Demo/inc --c99
-I./RTE/_sample
-Id:/ARM/PACK/ARM/CMSIS/5.8.0/Device/ARM/ARMSC000/Include
-Id:/Keil_v5/ARM/CMSIS/Include
-D__UVISION_VERSION="524" -DARMSC000
-o .\Obj\*.o --omf_browse .\Obj\*.crf --depend .\Obj\*.d`

### ASM:
`--cpu SC000 --li -g --apcs=interwork --pd "__MICROLIB SETA 1" 
-I .\RTE\_sample
-I d:\ARM\PACK\ARM\CMSIS\5.8.0\Device\ARM\ARMSC000\Include 
-I d:\Keil_v5\ARM\CMSIS\Include 
--pd "__UVISION_VERSION SETA 524" --pd "ARMSC000 SETA 1" --list ".\Lst\*.lst" --xref -o "*.o" --depend "*.d" 
`

### Linker:
`--cpu SC000 *.o 
--library_type=microlib --strict --scatter ".\Scatter\sample.sct" 

--entry=Reset --summary_stderr --info summarysizes --map --xref --callgraph --symbols 
--info sizes --info totals --info unused --info veneers 
 --list ".\Lst\sample.map" 
-o .\Obj\sample.axf 
`

github0null, 2021-12-19:
@ckforce 

`SC000` 和 `SC300` 已在 v2.15.3 中增加，具体见 changelog，帖子我先关闭了

closed 2021-12-19 by github0null