github0null/eide #82: 能提供CMSIS—DAP的烧录选项吗？

qihuanlu01, 2021-12-21:
能提供CMSIS—DAP的烧录选项吗？

github0null, 2021-12-23:
没什么问题我就关闭了

closed 2021-12-23 by github0null

wgco, 2022-03-11:
Traceback (most recent call last):
  File "C:\Users\wachw\AppData\Local\Programs\Python\Python39\Scripts\pyocd-script.py", line 33, in <module>
    sys.exit(load_entry_point('pyocd==0.0.0', 'console_scripts', 'pyocd')())
  File "C:\Users\wachw\AppData\Local\Programs\Python\Python39\Scripts\pyocd-script.py", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "C:\Users\wachw\AppData\Local\Programs\Python\Python39\lib\importlib\metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "C:\Users\wachw\AppData\Local\Programs\Python\Python39\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 972, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked  
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 627, in _load_backward_compatible
  File "<frozen zipimport>", line 259, in load_module
  File "C:\Users\wachw\AppData\Local\Programs\Python\Python39\lib\site-packages\pyocd-0.0.0-py3.9.egg\pyocd\__init__.py", line 21, in <module>
  File "<frozen zipimport>", line 259, in load_module
  File "C:\Users\wachw\AppData\Local\Programs\Python\Python39\lib\site-packages\pyocd-0.0.0-py3.9.egg\pyocd\gdbserver\__init__.py", line 17, in <module>
  File "<frozen zipimport>", line 259, in load_module
  File "C:\Users\wachw\AppData\Local\Programs\Python\Python39\lib\site-packages\pyocd-0.0.0-py3.9.egg\pyocd\gdbserver\gdbserver.py", line 41, in <module>
  File "<frozen zipimport>", line 259, in load_module
  File "C:\Users\wachw\AppData\Local\Programs\Python\Python39\lib\site-packages\pyocd-0.0.0-py3.9.egg\pyocd\rtos\__init__.py", line 19, in <module>
  File "<frozen zipimport>", line 259, in load_module
  File "C:\Users\wachw\AppData\Local\Programs\Python\Python39\lib\site-packages\pyocd-0.0.0-py3.9.egg\pyocd\rtos\argon.py", line 21, in <module>
  File "<frozen zipimport>", line 259, in load_module
  File "C:\Users\wachw\AppData\Local\Programs\Python\Python39\lib\site-packages\pyocd-0.0.0-py3.9.egg\pyocd\core\plugin.py", line 26, in <module>
ModuleNotFoundError: No module named 'pyocd._version'
终端进程“C:\WINDOWS\system32\cmd.exe /C pyocd flash --config u:\4、传感器系统\15、距离传感器\1.program\1.AJ-SR04M\Project\MDK-ARM\.eide\flash.pyocd.yaml -t cortex_m -f 4M u:\4、传感器系统\15、距离传感器\1.program\1.AJ-SR04M\Project\MDK-ARM\build\Flash\MDK-ARM.hex”已终止，退出代码: 1。

github0null, 2022-03-11:
@wgco 

你的 pyocd 没有安装好，pyocd 使用文档：https://docs.em-ide.com/#/zh-cn/download_to_device?id=%e4%bd%bf%e7%94%a8-pyocd-%e7%83%a7%e5%bd%95

> PS：我的意思是在论坛上开新帖子，这个 issue 跟你的问题没有关系