github0null/eide #88: F7编译没有输出了，版本2.16.4最新的，急！

ckforce, 2022-01-05:
> Executing task in folder X7-back-bootloader: & c:\Users\jiech\.vscode\extensions\cl.eide-2.16.4\bin\builder\mono.exe c:\Users\jiech\.vscode\extensions\cl.eide-2.16.4\bin\builder\unify_builder.exe -b d:\Keil_v5\ARM\ARMCC -M arm.v5.model.json -p f:\HONYAR\兰泽相关\产品\X7\X7-back-FM15L023UK6\X7-back-bootloader\build\FM15L023UK6\builder.params -m Fast-MULTHREAD <

终端将被任务重用，按任意键关闭。

ckforce, 2022-01-05:
> Executing task in folder test: & c:\Users\jiech\.vscode\extensions\cl.eide-2.16.4\bin\builder\mono.exe c:\Users\jiech\.vscode\extensions\cl.eide-2.16.4\bin\builder\unify_builder.exe -b d:\Keil_v5\ARM\ARMCC -M arm.v5.model.json -p c:\Users\jiech\Desktop\test\build\Debug\builder.params -m Fast-MULTHREAD <

终端将被任务重用，按任意键关闭。

github0null, 2022-01-05:
通过这个检查一下：https://discuss.em-ide.com/d/109-eide

ckforce, 2022-01-05:
[image]
按https://discuss.em-ide.com/d/109-eide检查都正常，还是没有任何输出

github0null, 2022-01-05:
你直接把整条命令放到 cmd 里运行，试试；

```
c:\Users\jiech.vscode\extensions\cl.eide-2.16.4\bin\builder\mono.exe c:\Users\jiech.vscode\extensions\cl.eide-2.16.4\bin\builder\unify_builder.exe -b d:\Keil_v5\ARM\ARMCC -M arm.v5.model.json -p f:\HONYAR\兰泽相关\产品\X7\X7-back-FM15L023UK6\X7-back-bootloader\build\FM15L023UK6\builder.params -m Fast-MULTHREAD
```

因为这个 `F7` 实际上就是将这条 cmd 发送到 vscode 终端去执行的

ckforce, 2022-01-05:
[image]

命令行是可以的，vscode里面不行。确切的应该说是powershell不行，cmd可以。把默认终端改成cmd就好了

closed 2022-01-05 by ckforce

github0null, 2022-01-05:
换到 cmd 虽然正常了，但实际上问题依然没解决；

对于 `powershell`，命令前会有一个 `&` 符号，看上面的日志这个命令的 `&` 号是有的，你可以放到系统的 `powershell` 再试试；

```
& c:\Users\jiech.vscode\extensions\cl.eide-2.16.4\bin\builder\mono.exe c:\Users\jiech.vscode\extensions\cl.eide-2.16.4\bin\builder\unify_builder.exe -b d:\Keil_v5\ARM\ARMCC -M arm.v5.model.json -p f:\HONYAR\兰泽相关\产品\X7\X7-back-FM15L023UK6\X7-back-bootloader\build\FM15L023UK6\builder.params -m Fast-MULTHREAD
```

ckforce, 2022-01-05:
加上&是可以的，然后我把terminal改回powershell又行了？？？这真是奇了怪了！

ckforce, 2022-01-05:
继续更新情况，改回来好了一会，大概1-2小时，然后又不行了，只能再改回cmd。。。