github0null/eide #90: 头文件出现“在 browse.path 中未找到包含文件。C/C++”

wangfreexx, 2022-01-13:
首先非常感谢你的付出
问题描述：
无法找到.h文件，但是能编译通过，vs版本1.63.2，eide版本2.16.5

github0null, 2022-01-13:
打开`输出面板`，切换到 `eide-cpptools-log`，里面会输出你打开过的源文件的 c/c++ 配置，你检查一下你的头文件是不是在相应的 `includePath` 或 `browsePath` 中

[image]

github0null, 2022-01-14:
波浪线提示的消息是 `未找到 main.h` 还是 `未找到 main.h 中的 xxx.h 或者 定义` ?

***

只要 eide 能够提供正确的配置，剩下的工作就靠 `C/C++` 插件了；

C/C++ 插件有一些命令可以重新扫描工作区，试试以下两种方法：

- `ctrl+shitf+p` 打开命令面板，输入：`Reset IntelliSense Database`，重置数据库，重启一下 vscode，等待 c/c++ 重新扫描整个项目

- `ctrl+shitf+p` 打开命令面板，输入：`Rescan Workspace`，可以立即扫描工作区

wangfreexx, 2022-01-15:
提示是在 browse.path 中未找到包含文件，但是我不知道browse.path在哪里设置

github0null, 2022-01-15:
@wangfreexx

eide 会提供所有的配置，在上面的 `eide-cpptools-log` 里搜索 `[folder]`, 就能找到 `browse.path` 配置，如下图：

[image]

wangfreexx, 2022-01-16:
图片folder提供了路径但是没有修改的地方
[image]
我尝试注释掉这个值，然后自己写include和browse，可以生效，看起来似乎是"configurationProvider": "cl.eide"没有生效？

github0null, 2022-01-16:
这些配置是根据项目信息生成的，是无法手动更改和添加的

只要 eide-cpptools-log 中有配置输出，说明 c/c++ 已经向 eide 请求了配置；

我记得 c/c++ 也会生成这种 log，相应的命令好像是：C/C++ Diagnose 之类的，你可以检查一下，看它实际收到的配置是什么。

wangfreexx, 2022-01-16:
[image]
看样子应该是收到了，但是还是没有，我很好奇其他人包括大佬的环境没有这个问题吗？我每次重新创建都会有这个问题

github0null, 2022-01-16:
那我就不清楚了

我这边肯定是没有这种问题的，检查你的 c/c++ 插件配置吧，或者重装一下 c/c++ 插件试试。

wangfreexx, 2022-01-17:
好的，麻烦了

bsbandwl, 2022-01-24:
> 好的，麻烦了

我以前遇到了 `eide-cpptools-log` 死活没有任何输出，最后不得不删了整个VSCode，然后纯净环境下，只安装C/C++等几个基本扩展，然后 `eide-cpptools-log` 就好了。ConfigurationProvider 也注册了。说明之前我装的二三十个小众扩展中，有和EIDE冲突的。

ckforce, 2022-01-31:
> 重装一下 c/c++ 插件

我也是前几天突然出现这个，重装c/c++插件之后ok了

TuRan-sino, 2022-02-05:
我发现如果我使用C_CPP1.7.1版本没有问题
更新至1.8.2版本就会出现报错

https://user-images.githubusercontent.com/51948125/152632713-68e13dc2-8989-435d-ad42-7d41075121d4.mp4

ckforce, 2022-02-05:
是的，更新了就会出现，我今天更新了一下，也发现了这个问题

github0null, 2022-02-06:
> 我发现如果我使用C_CPP1.7.1版本没有问题 更新至1.8.2版本就会出现报错 

参考：https://github.com/github0null/eide/issues/99#issuecomment-1030791515

wangfreexx, 2022-03-09:
[image]
终于我一个一个的新安装，重新写每一个语句，找到这条设置，当初不知道为什么加上去，但是注释掉后这个问题就没了，大佬可以关闭这个问题了

closed 2022-03-09 by github0null

Fluorine-fish, 2025-06-28:
可以直接将工作区根目录下的所有目录通过 文件->添加文件夹到工作区 来实现