github0null/eide #95: 对于不同Target的预定义宏，在编辑页面能否应用生效

bsbandwl, 2022-01-21:
问题大概是这样的：
我通过Keil的不同Target来生成不同的目标固件，而程序代码中不同的部分就靠宏来开关。这些宏绑定在不同Target中。
没有使用EIDE时，我在VSCode中，通过编辑c_cpp_properties.json，按照Target编写configurations，控制各个includePath和defines。
[image]
这样我切换不同configuration，实现不同的宏定义切换，保证看的代码正确显示。

用了EIDE后，虽然可以切换Target（我找了好一会儿才找到切换入口），也能查看不同Target的宏，但是VSCode里面显示的代码明显没有同步这个define和includePath。
希望能同步上，这样我完全用EIDE就行，不用自己再去创建configurations。
[image]

谢谢

github0null, 2022-01-21:
- 如果你使用 `eide`，**不要手动填写** `c_cpp_properties.json`，这个配置是固定不变的，始终为如下内容：
```
{
    "configurations": [
        {
            "name": "win32",
            "configurationProvider": "cl.eide"
        }
    ],
    "version": 4
}
```

- 对于不同的 `target`，预处理器定义都是独立的，你像 keil 那样去操作就行了

bsbandwl, 2022-01-21:
[image]

我把 c_cpp_properties.json 删除，然后重启软件，EIDE 重建了您上面这样的 c_cpp_properties.json。 确实使用了 cl.eide，但是宏还是没有生效

谢谢

github0null, 2022-01-21:
我这里是正常的，见下面 gif:

[image]

***

你可以试一试以下步骤：

- 打开 `输出面板`->`eide-cpptools-log` 检查 eide 提供的 `c_cpp_properties` 配置中是否存在该 `宏`

  [image]

- 如果上一步的 `c_cpp_properties` 配置没有问题，可能是 `C/C++` 插件没有及时刷新导致的，你可以打开命令面板输入以下命令重置数据库试试；

  [image]

bsbandwl, 2022-01-21:
我卸载了CMake插件，各种重启，都没在输出面板看到任何信息，只有 eide-log 有输出
[image]
[image]
[image]
发现确确实实没有输出：
[image]
但是C/C++有警告：
[2022/1/21下午6:07:52] 对于 C 源文件，cStandard 已根据编译器参数和查询 compilerPath 从“c99”更改为“c17”:“D:/Program Files (x86)/Microsoft Visual Studio/2019/Professional/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe”
[image]

> 我这里是正常的，见下面 gif:
> 
> [image]
> 
> 你可以试一试以下步骤：
> 
>     * 打开 `输出面板`->`eide-cpptools-log` 检查 eide 提供的 `c_cpp_properties` 配置中是否存在该 `宏`
>       [image]
> 
>     * 如果上一步的 `c_cpp_properties` 配置没有问题，可能是 `C/C++` 插件没有及时刷新导致的，你可以打开命令面板输入以下命令重置数据库，然后重启试试；
>       [image]

github0null, 2022-01-21:
只要打开了 eide 项目，就会注册 `configurationProvider`，无论注册成功与否，都会存在输出

[image]

[image]

***

> 如果没有使用 eide 提供 c/c++ 配置，代码中的 `#include"xxx.h"` 指令将存在红色波浪线，同时跳转定义等功能也将无法正确跳转至相应位置，如果一切正常，那就没问题，你可以检查一下；

[image]

***

建议先删除整个 eide 安装目录（`C:\Users\<用户名>\.vscode\extensions\cl.eide-2.16.6`）后，重装插件。

如果还不能解决，我也没有好的办法

bsbandwl, 2022-01-24:
各种折腾之后，删除了整个VSCode，然后重装，只安装C/C++扩展，之后按照EIDE，发现功能正常。说明之前我大量安装的插件中，有一个产生了问题

closed 2022-01-24 by bsbandwl

github0null, 2022-01-24:
好吧，论坛里也有一个跟你一样，目前不知道什么原因导致的

> 我也检查了一下，除非没有收到 `project.opened` 事件，否则这个一定会产生输出的；而 `project.opened` 事件在项目被打开后就会被立即发出

TuRan-sino, 2022-01-29:
我也产生了和author一样的问题, 
EIDE还提示了这个信息
[image]