github0null/eide #98: Warning: parse cpp configuration error!

TuRan-sino, 2022-01-29:
<b>解析C/C++配置的时候发生错误</b><br>

标题是VsCode打开时给我报的错误, 看样子应该是C/C++解析不了本插件提供的参数<br>具体表现就是
1.  所有的函数和头文件转不到定义, 显示 `no definition founded`
2.  C/C++提供的`PROBLEMS`一直显示 "无法打开源文件 ......."
3. `output` 里面的 `eide-cpptools-log`显示空白, 没有任何内容
4. 每次删除插件重新安装之后不关闭VSCODE的前提下继续使用插件没有这个问题.  **但是, 在重启vscode之后这个问题又会重新回来**
5. 删除插件第一次安装时现`eide-cpptools-log`是有内容的, 显示内容如下
```
[init] register CustomConfigurationProvider done !

[folder] provideFolderBrowseConfiguration for 'd:\Work\Project\Graduation\KEIL'
browsePath: []
compilerPath: null
compilerArgs: null

[source] provideConfigurations
- uri:
    $mid: 1
    fsPath: d:\Work\Project\Graduation\USER\main.c
    _sep: 1
    path: /D:/Work/Project/Graduation/USER/main.c
    scheme: file
  configuration:
    standard: c99
    includePath:
      - d:/Work/Project/Graduation/CORE
      - d:/Work/Project/Graduation/USER
      - d:/Work/Project/Graduation/FWLIB/inc
      - d:/Work/Project/Graduation/HARDWARE
      - d:/Work/Project/Graduation/SYSTEM
      - d:/Work/Project/Graduation/KEIL/.eide/deps
      - D:/Keil_v5/ARM/ARMCC/include
      - D:/Keil_v5/ARM/ARMCC/include/rw
    defines:
      - __VSCODE_CPPTOOL
      - STM32F10X_HD
      - USE_STDPERIPH_DRIVER
    forcedInclude:
      - c:\Users\13275\.vscode\extensions\cl.eide-2.16.6\bin\include\internal_headers\armcc_intr.h
    compilerPath: null
    compilerArgs: null

[folder] provideFolderBrowseConfiguration for 'd:\Work\Project\Graduation\KEIL'
browsePath:
  - d:/Work/Project/Graduation/USER/*
  - d:/Work/Project/Graduation/SYSTEM/*
  - d:/Work/Project/Graduation/HARDWARE/*
  - d:/Work/Project/Graduation/FWLIB/src/*
  - d:/Work/Project/Graduation/CORE/*
compilerPath: null
compilerArgs: null

[source] provideConfigurations
- uri:
    $mid: 1
    fsPath: d:\Work\Project\Graduation\USER\main.c
    _sep: 1
    path: /D:/Work/Project/Graduation/USER/main.c
    scheme: file
  configuration:
    standard: c99
    includePath:
      - d:/Work/Project/Graduation/CORE
      - d:/Work/Project/Graduation/USER
      - d:/Work/Project/Graduation/FWLIB/inc
      - d:/Work/Project/Graduation/HARDWARE
      - d:/Work/Project/Graduation/SYSTEM
      - d:/Work/Project/Graduation/KEIL/.eide/deps
      - D:/Keil_v5/ARM/ARMCC/include
      - D:/Keil_v5/ARM/ARMCC/include/rw
    defines:
      - __VSCODE_CPPTOOL
      - STM32F10X_HD
      - USE_STDPERIPH_DRIVER
    forcedInclude:
      - c:\Users\13275\.vscode\extensions\cl.eide-2.16.6\bin\include\internal_headers\armcc_intr.h
    compilerPath: null
    compilerArgs: null
```
6. `c_cpp_propertise.json`显示内容如下<br>
[image]
7. 重启之后vscode报错如下<br>
```
Warning: parse cpp configuration error!
```

github0null, 2022-01-30:
检查你的 code-workspace 文件的内容是否正确，以及字符集编码，如果该文件不是正常的 json，才会出现 `parse 
xxx error` 错误

TuRan-sino, 2022-01-30:
> 检查你的 c_cpp_properties.json 文件的内容是否正确，以及字符集编码，如果该文件不是正常的 json，才会出现 `parse xxx error` 错误

文件编码位UTF-8
json文件并没有出现任何格式错误, 至少从vscode提供的报错来看没有
json文件如下
```json
{
    "configurations": [
        {
            "name": "win32",
            "configurationProvider": "cl.eide"
        }
    ],
    "version": 4
}
```

github0null, 2022-01-30:
出现该警告的只有这一处地方，只有该文件内容不是一个正常的 json 字符串时，才会有这个异常

```js
protected Parse(jsonStr: string): CppConfig {
        try {
            return <CppConfig>jsonc.parse(jsonStr);
        } catch (error) {
            GlobalEvent.emit('msg', newMessage('Warning', 'parse cpp configuration error !'));
            return this.GetDefault();
        }
    }
```

***

如果 eide 在读取该文件的时候，刚好有其他插件正在写入该文件，可能会导致读取的文件内容不完整，所以有这个异常，你可以先**卸载掉其他没用的插件**，再试试

TuRan-sino, 2022-01-30:
好的, 我先尝试一下

TuRan-sino, 2022-02-02:
重新安装了一遍C/C++拓展就好能够成功运行了
多谢作者再过年期间帮助我解决问题, 新年快乐

closed 2022-02-02 by TuRan-sino