github0null/eide #102: 使用pyocd烧录时存在问题。

liaohongxing, 2022-03-02:
```
pyocd list --pack .\ArteryTek.AT32F413_DFP.2.0.3.pack --targets  
// 似乎 pyocd 对特殊字符有处理，采用第一列的值
  _at32f413c8t7             ArteryTek               -AT32F413C8T7                AT32F41x Series, AT32F413   pack
  _at32f413cbt7             ArteryTek               -AT32F413CBT7                AT32F41x Series, AT32F413   pack
  _at32f413cbu7             ArteryTek               -AT32F413CBU7                AT32F41x Series, AT32F413   pack
  _at32f413cct7             ArteryTek               -AT32F413CCT7                AT32F41x Series, AT32F413   pack
  _at32f413ccu7             ArteryTek               -AT32F413CCU7                AT32F41x Series, AT32F413   pack
  _at32f413kbu7_4           ArteryTek               -AT32F413KBU7-4              AT32F41x Series, AT32F413   pack
  _at32f413kcu7_4           ArteryTek               -AT32F413KCU7-4              AT32F41x Series, AT32F413   pack
  _at32f413rbt7             ArteryTek               -AT32F413RBT7                AT32F41x Series, AT32F413   pack
  _at32f413rct7             ArteryTek               -AT32F413RCT7                AT32F41x Series, AT32F413   pack
  _at32f413tbu7             ArteryTek               -AT32F413TBU7                AT32F41x Series, AT32F413   pack
  _at32febkc8t7             ArteryTek               -AT32FEBKC8T7                AT32F41x Series, AT32F413   pack
```

```
pyocd flash --pack .\ArteryTek.AT32F413_DFP.2.0.3.pack -t "_at32f413cbt7" x.bin
0000972 C No ACK received [__main__]  // 有反应, 但是因为我没插daplink 提示ACK接受不到
```

```
pyocd flash --pack .\ArteryTek.AT32F413_DFP.2.0.3.pack -t "-AT32F413C8T7" x.bin
pyocd load: error: argument -t/--target: expected one argument // 提示参数错误,烧录失败
```

测试的pack已上传 ，可以测试下

[image]
[image]
[image]
[image]

[ArteryTek.AT32F413_DFP.2.0.3.zip](https://github.com/github0null/eide/files/8171424/ArteryTek.AT32F413_DFP.2.0.3.zip)

github0null, 2022-03-03:
`-t` 参数是指 `target type`，因为 `part_number` 是较为完整的名称，所以我用的就是这个字段，并不是 `name`，见下方官方的文档：

[image]

有些时候，`name` 只是一个系列名，而不是准确的芯片名，如果这时用 name，可能会出现问题

[image]

***

你这个问题，就是 pack 包没做好的原因，因为这个包里的芯片名称就是这样的（这个 pack 包本质就是一个压缩包，pyocd 靠读取里面的 `pdsc` 文件来获取芯片信息），而 pyocd 不允许在名称开头带有 `-`，因为 `-` 代表一个命令的开始

[image]

你可以先把 `pdsc` 文件里的 Device name 中的 `-` 去掉，然后压缩成 pack，再试试

liaohongxing, 2022-03-03:
能不能更改一下流程，切换到 pyocd 时， “目标芯片名称”  这个input框允许手动输入名称， 比如输入 _at32f413cbt7 ，
不强行验证 合法性(或者提醒一下然后继续) ，避免越来越多国内杂牌厂商命名不规范问题。

github0null, 2022-03-03:
后面会改进一下的，目前的话，可以先把 pyocd 的下载命令复制出来，用自定义的 shell 命令烧录方式去执行

github0null, 2022-03-07:
已在新版本中作了调整，先关闭了

closed 2022-03-07 by github0null