github0null/eide #104: armclang: error:cannot specify -o when generating multiple output files

WeNeedChinese, 2022-03-11:
编译的时候出现
armclang: error:cannot specify -o when generating multiple output files
错误

github0null, 2022-03-11:
检查你的编译参数里是否有多余的值

WeNeedChinese, 2022-03-11:
> 检查你的编译参数里是否有多余的值

请问编译参数在哪？有没有相关的资料推荐学习一下？我没有学习过编译原理。

github0null, 2022-03-11:
[image]

***

如果编译选项里有不正确的参数的话，可能会导致这种情况

你打开 **构建配置**->**编译器选项** -> **C/C++** 检查一下

WeNeedChinese, 2022-03-11:
> [image]
> 
> 如果编译选项里有不正确的参数的话，可能会导致这种情况
> 
> 你打开 **构建配置**->**编译器选项** -> **C/C++** 检查一下

[image]
我检查了我的编译器选项（从keil复制过来的），最后一条命令的确出现-o了，请问这个语句应该怎么修改才能在EIDE上编译？这个语句是什么意思？

github0null, 2022-03-11:
只留下 `-f` 和 `-w` 开头的参数，其他的全删掉

文档里有描述如何导入项目，https://docs.em-ide.com/#/zh-cn/import_project

Keil 上的勾选框选项，eide 也都有，需要复制的，只有 keil `misc-control` 里的东西

WeNeedChinese, 2022-03-11:
> 只留下 `-f` 和 `-w` 开头的参数，其他的全删掉
> 
> 文档里有描述如何导入项目，https://docs.em-ide.com/#/zh-cn/import_project
> 
> Keil 上的带有勾选框的选项，eide 也都有，需要复制的只有 keil `misc-control` 里的东西

[image]
[image]
已经按照教程导入工程和设置相关选项，还是无法在EIDE里面编译，但可以在Keil里面编译。请问keil里面C/C++选项卡下面三行与EIDE下面三行是如何对应的？

github0null, 2022-03-11:
- 先把这些勾选框，在 eide 的 C/C++ 页面里勾上，没有的先不管

  [image]

- `Misc-Controls` 里的填到 `编译器附加选项(全局) ` 里

  [image]

  还剩两个框，因为有时候 C 和 C++ 的一些选项是不兼容的，这时候要分开填在 `C编译器附加选项` 和 `C++编译器附加选项` 里面

WeNeedChinese, 2022-03-11:
> * 先把这些勾选框，在 eide 的 C/C++ 页面里勾上，没有的先不管
>   [image]
> * `Misc-Controls` 里的填到 `编译器附加选项(全局) ` 里
>   [image]
>   还剩两个框，因为有时候 C 和 C++ 的一些选项是不兼容的，这时候要分开填在 `C编译器附加选项` 和 `C++编译器附加选项` 里面（导入的 Keil 项目一般不用管这个）

在按要求重新导入之后出现以下问题：
[image]

请问是不是因为没有下载芯片支持包或者我导入芯片支持包的位置不对导致这个宏定义没有开启，出现编译错误。

github0null, 2022-03-11:
检查一下 `宏定义` 是否导入过来了，没有的话，手动加上，你之前发的参数中是有 `ARMCM4` 宏的

[image]

WeNeedChinese, 2022-03-11:
> 检查一下 `宏定义` 是否导入过来了，没有的话，手动加上，你之前发的参数中是有 `ARMCM4` 宏的
> 
> [image]

调试了一个下午之后终于编译成功了，总结以下就是项目属性文件要加上芯片型号，并且把没有成功导入的文件导入到原来的位置，构建器选项也要一一对应，感谢大佬的耐心指导！

closed 2022-03-11 by github0null