github0null/eide #106: MDK ThreadX项目编译失败

CaiHua1995, 2022-03-19:
STM32 CubeMX 生成的普通项目然后自己移植了threadx，在mdk上可以编译通过，在vscode编译提示threadx内部的汇编文件编译错误，在构建器里面将汇编器类型改为asm-clang后threadx内部汇编文件可以编译通过，但是STM32启动文件又编译失败。整个mdk项目放到了附件里面

CaiHua1995, 2022-03-19:
链接：https://pan.baidu.com/s/1-jRpD67GRT8jsrfI21FdYQ 
提取码：vctc
这个是整个项目文件我把它放在了百度云盘

github0null, 2022-03-19:
这个插件是不能为单个源文件设置用什么编译器的，只能全局设置；threadx 汇编是 gcc 的，你的 cubemx startupxx.s 用的 armcc 汇编，

CaiHua1995, 2022-03-21:
好吧！threadx 汇编是armcc v6 但是cubemx startupxx.s 好像还是老版本的！在MDK里面设置成 v6编译器编译是没问题的

github0null, 2022-03-21:
你可以把先把 `startup_stm32f407xx.s` 用 armcc5 编译成 `.o`，添加到项目里，然后直接用 armcc6 编就行了

[image]

CaiHua1995, 2022-03-21:
好的！谢谢

closed 2022-03-21 by CaiHua1995

TP-Thread, 2024-04-18:
可以在EIDE里添加arm-clang(arm syntax)吗？
[image]

jmdweem, 2025-04-04:
> 可以在EIDE里添加arm-clang(arm syntax)吗？ [image]

同样希望加入Arm Syntax的选项