github0null/eide #109: 升级3.1以后，没有 'c_cpp_properties.json' 文件，切换Target但是编辑界面宏定义不显示

bsbandwl, 2022-04-06:
升级3.1以后，没有 'c_cpp_properties.json' 文件，切换Target但是编辑界面宏定义不显示。

原本切换不同Target，代码还能够自动显示正确的宏定义。现在没 'c_cpp_properties.json' 文件，所有的宏定义都没了，都是未定义状态，但是编译还能编译正确

github0null, 2022-04-06:
目前通过 cpptools API 在内部来提供代码跳转信息，已经**不再**通过写入 `c_cpp_properties.json` 这个文件来提供配置了，因此这个文件已经被移除了

当你打开工作区时，eide 会注册配置提供者，C/C++ 插件会弹出提示，必须选择 `Allow`，eide 才会提供配置

[image]

打开输出面板，可以看到 eide 提供的配置内容，这就是之前 `c_cpp_properties.json` 里的内容

[image]

bsbandwl, 2022-04-06:
C/C++ 插件没提示我，怎么办？在哪手动配置这个allow

github0null, 2022-04-06:
`ctrl+shift+p` 打开命令面板，输入 `Change Config`，可以进行更改

[image]

bsbandwl, 2022-04-06:
搞定！THX

closed 2022-04-06 by bsbandwl