github0null/eide #114: Gets a Trojan Horse Warning from Antivirus

vishnumaiea, 2022-04-20:
Installed this extension and suddenly I got a threat warning from Avast. It complains that **memap.exe** is infected by some malware.

[image]

github0null, 2022-04-20:
@vishnumaiea 

`memap.exe` is build by PyInstaller.  This may be a false positive. 

https://stackoverflow.com/questions/43777106/program-made-with-pyinstaller-now-seen-as-a-trojan-horse-by-avg

This program used to parse *.map files. like this:

[image]

vishnumaiea, 2022-04-20:
I have reported a false positive.

closed 2022-04-20 by github0null