github0null/eide #143: 如何配置程序存储器的代码偏移？

Antecer, 2022-07-17:
如图，这是Keil的配置选项：
[image]

需求是将编译后的程序代码放入指定的ROM地址段。

github0null, 2022-07-17:
这个些选项，Keil 会生成 链接参数，将右图中的 `CODE(...) ...` 这些参数复制到 eide 的 构建配置 -> Linker 配置页面 -> `Other Linker Options` 中就行了

[image]

Antecer, 2022-07-17:
[image]
[image]
链接器参数拷贝过来，编译出错。

github0null, 2022-07-17:
去掉 逗号 ',' 再试

Antecer, 2022-07-17:
[image]
还是不行

github0null, 2022-07-17:
刚看了一下，Keil 做了预处理的，你要先输出 `.lnp` 参数文件，然后再看

勾上 `output BAT` 之后，点编译，会生成 `.lnp` 文件到工程目录下

[image]

Antecer, 2022-07-17:
成功了，用 CLASSES() 把参数都包括起来就正确了。

closed 2022-07-17 by Antecer