github0null/eide #145: chip seems get burned while debugging in vscode

TheSnowfield, 2022-07-28:
I knew that sounds weird, but this is the things what Im experiencing.

While I clicked the "debug" button of the vscode, it works fine. then I click "step", chip reseted and goes to the head of the program.
I clicked "step" many times but got the same thing, I give up and cancel the debug.

THEN THE CHIP BURNED.

I tried to use keil to debug it (Keil is always works fine), then got the error.
```
RDDI-DAP Error
Error: Flash Erase failed  -  Target DLL has been cancelled
```
I tried to use openocd, connect and try to erase the chip. it can not detect any chips, just like connect nothing. even I use a new cmsis-dap.
```
GNU MCU Eclipse OpenOCD, 32-bitOpen On-Chip Debugger 0.10.0+dev-00593-g23ad80df4 (2019-04-22-16:40)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: JTAG Supported
Info : CMSIS-DAP: FW Version = 0254
Info : CMSIS-DAP: Serial# = 0700000105deff3834334e4643206540a5a5a5a597969908
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 2000 kHz

```

I have checked swdclk and swdio and gnd wires, no dummy soldering or line broken.
Im confidence in my circuit design correctly, cuz before debugging in vscode, it works fine and debugged in Keil many times.

Currently, Im going to replace the chip.

TheSnowfield, 2022-07-28:
After replace a new chip, I can got chip running normally.
Idk what the happend, from start to the end.

github0null, 2022-07-29:
所以问题的原因是什么？

据我所知，烧掉芯片绝大多数都是接错了电源，或者超过引脚电压限制

TheSnowfield, 2022-07-29:
> 所以问题的原因是什么？
> 
> 据我所知，烧掉芯片绝大多数都是接错了电源，或者超过引脚电压限制

是的, 所以我也覺得非常疑惑.
但是原因尚未查明. 我在調試過程中沒有插拔swd的動作, 僅僅只是點擊鼠標, 所以不太可能是由於ESD造成.

隨後的調查中發現, 代碼存在bug. 貌似是往0x00000000寫了一段數據, 可能覆蓋了一些中斷表引發上文所述的"調試時崩潰並自動reset"的情況, 但我不是很相信它會造成晶片的物理性損壞.

我有點開始懷疑cmsis-dap的生產廠商. 但鑒於這是調試中的偶發現象且此前從未發生過, 更沒有提前準備好設備去應對, 所以也就無從得知了.

github0null, 2022-07-30:
如果无法确定问题是由  `该插件内置的烧录软件` 导致，我将关闭该 issue

closed 2022-07-30 by github0null