github0null/eide #152: feature request about importing keil project

TheSnowfield, 2022-08-14:
I found EIDE always ignores the `Misc controls` option of the arm compiler and linker while importing the keil project.
[image]
[image]

they can cause fatal compiling or linking errors in some cases. 
now I can add them to eide builder options manually but I have cost the whole night tracking the issue.

I tried to fork and create a PR, but I cannot deal with the code. I guess the Keil XML parser seems cannot pass the extra arguments to the build options, they're the two independent parts.

so the feature request comes. :3

github0null, 2022-08-14:
这个文档里本来就有说明：
https://docs.em-ide.com/#/zh-cn/import_project

TheSnowfield, 2022-08-14:
> 这个文档里本来就有说明： https://docs.em-ide.com/#/zh-cn/import_project

doc確實沒什麽問題, 但是導入工程的時候這些字段完全是可以被解析的,
並且這些選項在 build option 中也可以被手動設置, 導入的時候故意丟掉這些字段, 個人感覺有一點不符合邏輯。
倘若只是因爲架構問題暫時不好去實現它, 那好像也沒什麽別的辦法了。

closed 2022-08-14 by TheSnowfield