github0null/eide #161: Unable to build the example with STM8S103 and SDCC for debug

mkSaBtsy, 2022-09-03:
After discovering EIDE in Visual Studio Code I want to debug STM8 microcontroller with ST-LINK. I'm using SDCC and I have already successfully built hex-files and flashed it to the uC. But I need to debug them as well.

Unfortunately I'm not even able to build the example for the STM8S103 CPU in debug configuration. I instantly get the error **"Invalid '$use' option!,please check compile option 'linker.$use'"**

I found the option "$use" in linker parameters in the file builder.params. So I triedt to delete the '$use' option because it is not recognized by SDCC. Alas, this file seems to be auto generated. After I attemp to build the configuration the '$use' parameter is added agin to the linker parameters.

Does anyone have a suggestion for me?

github0null, 2022-09-04:
I have fixed it.

Restart VSCode, eide will fetch update for `eide-binaries` and update to `8.3.4`

mkSaBtsy, 2022-09-04:
Dear github0null,

Great thanks for the quick fix. 
Now I can at least build the debug configuration. However, the executing path for the "sdcc fix elf" task seems to be wrong in the template. It only work if I manually correct the path to the "stm8-objcopy.exe" that I searched in the .eide folder. Maybe the template has to be adapted.

Not working preset in the template:
[image]
Output in terminal:
[image]

Workaround that worked for me:
[image]
Output:
[image]

Unfortunateli I'm still unable to actually debug my chip. Although I put the openocd - .cfg- files for my chip in every folder mentioned there: 
https://em-ide.com/docs/modules/flasher#specific-configurations 
I don't get openocd to choose as "flasher-configuration". See here:
[image]
I even tried to put separately downloaded binaries for openocd into the folder configured in the eide plugin settings.
Can you give me a hint please?

github0null, 2022-09-04:
This template is maybe too old, some env variables are deprecated

You need remove `${ExeDir}` env, only use `stm8-objcopy.exe`, because this exe is already in system path.

[image]

github0null, 2022-09-04:
There are all available env variables:

https://em-ide.com/docs/modules/builder#user-task-variables

github0null, 2022-09-04:
>Unfortunateli I'm still unable to actually debug my chip. 

This plug-in can't debug your chip, it's only can `build`, `program flash`

You need other debugger plug-ins

github0null, 2022-09-04:
> I don't get openocd to choose as "flasher-configuration". See here:

Now we not support openocd for `8BIT Project` !
[image]

You can use any shell command to flash your board by `Shell Flasher` !
[image]

mkSaBtsy, 2022-09-04:
> Now we not support openocd for 8BIT Project !

Ah, I see. Indeed, thanks to your plugin I'm able to flash STM8-chips in a comfortable way.

> This plug-in can't debug your chip, it's only can build, program flash

Since your answer I'm trying to get the STM8-debug plugin to work. 

Thanks a lot for your explanations.

closed 2022-09-04 by mkSaBtsy