github0null/eide #162: output debug symbols for STM8

mkSaBtsy, 2022-09-10:
I'm still struggeling with debugging a STM8S target. It seems to me that eide doesn't pass the proper arguments to sdcc for generating debug symbols. The STM8 debug plugin complains about DWARF error:

[image]

If I compile the program manually by entering the commands into the terminal the debug plugin works like expected.

[image]
[image]

Unfortunately I sill haven't found out how to output the exact commands and arguments that are used when compiling.

github0null, 2022-09-11:
Maybe you need apply `--debug` params to `Builder Options` -> `Global Options` -> `Other Global Options`

I'll fix this later

mkSaBtsy, 2022-09-11:
> ... apply `--debug` params to `Builder Options` -> `Global Options` -> `Other Global Options `

I've tryed that too. Doesn't work either.

>I'll fix this later
So there is something to fix? No need for hurry. I'm glad if I can help improving.
In the meantime I'm still able to build by cmd.

github0null, 2022-09-13:
> So there is something to fix? No need for hurry. I'm glad if I can help improving.
In the meantime I'm still able to build by cmd.

Because i found that `output debug info` options was ignored by builder, so `--debug` args will be not passed to sdcc

I have fixed this in v3.9.0

You can use `Show Compiler Commadline` menu item to check your compiler base args

[image]

[image]

mkSaBtsy, 2022-09-13:
>You can use Show Compiler Commadline menu item to check your compiler base args

I have been aware of the `Show Compiler CommandLine` entry of the context menu. However, bexause debug info was not put out although the `--debug` option was shown in the supposed `Compiler CommandLine`. So I guessed that the compiler / linker was not invoked with the commands that the output of this menu item claims to.

>I have fixed this in v3.9.0

Humble question: Are you sure? The STM8 debugger plugin is still complaining about DWARF error.
[image]

When I manally enter the commands everythins seems shiny.
[image]

github0null, 2022-09-14:
> Dwarf Error: xxxxx 

This is an unusual mistake

When i use sdcc v4.2.0, it tell me this error

But debugging information already exists in ELF

[image]

When i use sdcc v4.0.0, it works

[image]

---

I think the SDCC has some errors in processing debugging information in lib files.

In eide builder, after compiled all c source files, eide will make a lib file from all .rel file (not contain main.rel) 

And then, eide will use a command like `sdcc main.rel no_entry_bundled.rel` to link your elf/hex file

Because this operation will help sdcc reduce image size, ref: https://sourceforge.net/p/sdcc/discussion/1865/thread/e395ff7a42/#a03e

mkSaBtsy, 2022-09-17:
>When i use sdcc v4.0.0, it works

Yeah, I can confirm that the debugger plugin doesn't complain about "DWARF error" when I use SDCC v4.0.0. However, v4.0..0 seems to generate different assembly code too. After switching back I was not even able to get my super simple "blinky" program to run with release config. The debugger plugin doesn't work too. With v4.2.0 the LED is blinking when built in "release" config and manually compiled I'm able to debug the code, at least with manually started openocd (see [here](https://github.com/github0null/stm8-debug/issues/13#issuecomment-1245901744)). So using the outdated compiler is not satisfying at all. Attached both .lst files from my blinky program:
[lst_comparison.zip](https://github.com/github0null/eide/files/9591247/lst_comparison.zip)

>And then, eide will use a command like sdcc main.rel no_entry_bundled.lib to link your elf/hex file

Is it possible to make a switch for activating and deactivating this feature? At least for my simple blinky program the .elf file generated by eide is some bytes bigger than that one I compiled manually. See below:
[elf_comparison.zip](https://github.com/github0null/eide/files/9591249/elf_comparison.zip)

github0null, 2022-09-17:
> After switching back I was not even able to get my super simple "blinky" program to run with release config.

I can sure that compiler have no problems, because this plug-in was use v4.0.0 at the beginning

> Is it possible to make a switch for activating and deactivating this feature?

Sorry, I have no such plans in the near future

I think debug feature is not very necessary for sdcc, because sdcc does not well in debug.

And the `stm8-debug` gdb debugger is from stvp, sdcc not have a official gdb

mkSaBtsy, 2022-09-17:
>I can sure that compiler have no problems, because this plug-in was use v4.0.0 at the beginning

I've the impression that the additional label(?) `__sdcc_gs_init_startup:` in line 59 is responsible for my problem. It is only generated by v4.0.0 not by v4.2.0. 
Unfortunately assembly is not my best known programming language. So I'm not able to determine what else is going on exactly in the .lst files.

>Sorry, I have no such plans in the near future
>I think debug feature is not very necessary for sdcc, because sdcc does not well in debug.
 
I see and I accept that. 

Then do you have a hint for me, why the same, yet very simple, code doesn't work in v4.0.0?
[Stm8_3.zip](https://github.com/github0null/eide/files/9591352/Stm8_3.zip)
Otherwise you can close this issue.

Thanks.

closed 2022-09-17 by github0null