github0null/eide #166: stcgal: command not found

dzylikecode, 2022-09-17:
Actually, I can run `stcgl` in `conda` environment. However, I don't know how to configure the task. I have tried to modify the flash task, but it doesn't work. It still executes like below:
```txt
终端进程“/bin/bash '-c', 'stcgal  -P auto -b 115200 -p /dev/ttyUSB0 "/mnt/f/Git_WorkSpace/MCU/MCU-8051/test2/build/Debug/test2.hex"'”启动失败(退出代码: 127)。 
```
By the way, I use `zsh` instead of bash.

github0null, 2022-09-17:
插件运行在 vscode 环境中，所有环境变量继承自 vscode

先在 vscode 终端中运行 stcgal，如果找不到，说明你的系统环境变量 PATH 不对

如果你确定你的系统中能找到 stcgal，在系统 shell 中执行一次 `code .` 刷新系统变量

dzylikecode, 2022-09-17:
我是用了`alias stcgal="/home/minus/miniconda3/envs/mofan/bin/stcgal"`, 但是不作用

dzylikecode, 2022-09-17:
我在bash命令输入这个烧录的命令, 可以运行, 但是烧不进去, 它一直卡在这个地方

[image]

这是什么原因呢, 是不支持这个芯片吗

我的芯片是 STC89C516

github0null, 2022-09-17:
> 我是用了alias stcgal="/home/minus/miniconda3/envs/mofan/bin/stcgal", 但是不作用

插件使用的命令是固定的：`bash -c "stcgal xxxx"`，它就是在找 stcgal 这个二进制文件 ！！，不论你要使用何种 linux 的花活儿，你必须确保使用 /bin/bash 能够找到这个文件或命令，否则，请将 stcgal 设置到系统变量 PATH 中

> 可以运行, 但是烧不进去, 它一直卡在这个地方

请先阅读文档：https://em-ide.com/zh-cn/docs/modules/flasher#stcgal

dzylikecode, 2022-09-17:
好的,明白了, 非常感谢 @github0null

closed 2022-09-17 by dzylikecode