github0null/eide #167: Map View broken on Ubuntu 18

AAsyunkin-se, 2022-09-17:
src\extension.ts has 'python memap ....' hardcoded. This launches python 2.7 on Ubuntus before Focal.
Tried setting alias for python3 but that didn't work (and also not a great workaround with pyenv installed).

I realise Ubuntu 18 LTS is almost out of support, but should be a quick fix anyway?

Thanks

github0null, 2022-09-17:
Python3 is ok !  This tool supports python3.

Make sure the environment variables are correct so that the plug-in can find Python commands.

If a message is displayed indicating that the relevant package is missing, install it manually.

[image]

AAsyunkin-se, 2022-09-17:
Soft link is not a solution, I am afraid. It will break a) dependencies of other packages, either at installation or at the runtime and b)   https://legacy.python.org/dev/peps/pep-0394/ 
I would propose two different ways to solve the problem.
Either:
1) Allow user to specify which python he needs to use (same way as which GCC etc)
or
2) Adopt the same approach as you have for cppcheck - it tries to invoke python3 and, if not found, reverts to python.

github0null, 2022-09-18:
The tool supports python2 and python3, so I started with the `python XXX` command.

I don't mean to break your package dependencies or your Python environment.

But, at least, your system should have a valid `python` command, isn't it ?

AAsyunkin-se, 2022-09-18:
Apologies, I had an issue with python2 setuptools. I can confirm python2 works correctly too, after installing all missing modules. Thanks for your help, anyway!

closed 2022-09-18 by AAsyunkin-se