github0null/eide #172: [bug] I just want to give a suggestion

Boy-lzm, 2022-09-29:
[image]
For a more concise interface, can you handle the red circle? For example, when my mouse moves to the source file, the red circle will appear instead of always showing**What are you doing?**
Tell us what you were doing when the bug occurred.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows 10 xxx]
 - EIDE Version [e.g. v3.8.3]
 - VSCode Version [e.g. v1.68.1]

**Additional context**
some logs, like: `<user_home>/.eide/cl.eide.log`

github0null, 2022-09-29:
这只是个按钮，用来排除该文件

不清楚你要表达什么

Boy-lzm, 2022-09-29:
> 

> 这只是个操作按钮，用来解决该文件
> 
> 的按钮只是其中一个，当有这样的条件，这样的时候，它就会出现
> 
> 你要表达什

closed 2022-09-29 by Boy-lzm

Boy-lzm, 2022-09-29:
> 这只是个操作按钮，用来排除该文件
> 
> 这样的按钮有很多，这只是其中之一，当某些条件合适时，它就会显示出来
> 
> 不清楚你要表达什么

您好，我的意思是这个排除文件的按键能不能不是一直显示，只有鼠标移动到目录中的文件时才显示。