github0null/eide #175: [question] --print-memory-usage cause linker complains

TheSnowfield, 2022-10-11:
**What you want to do**
try using a compiler that is not included in EIDE yet to build the project.

**Describe the problem**
`--print-memory-usage` causes linker complaints.

```
[ INFO ] start linking ...
tc32-elf/bin/ld.exe: unrecognized option '--print-memory-usage'
tc32-elf/bin/ld.exe: use the --help option for usage information
collect2: ld returned 1 exit status
```

**Expected behavior**
Pass the compilation.

**To improve**
hope EIDE can offer an option to disable the `--print-memory-usage` default behavior, 
cuz some old linkers do not support this command.

currently using GCC 4.
```
Using built-in specs.
COLLECT_GCC=opt/tc32/bin/tc32-elf-gcc.exe
Target: tc32-elf
Configured with: ../../gcc-4.5.1/configure --target=tc32-elf --prefix=/opt/tc32 --disable-nls --disable-shared --disable-threads --with-gcc --with-gnu-ld --with-gnu-as --with-dwarf2 --enable-languages=c --disable-libstdcxx-pch --disable-libmudflap --disable-libgcc --disable-libssp --disable-multilib --disable-bootstrap --with-mpc=/opt/addontools --with-mpfr=/opt/addontools --with-gmp=/opt/addontools
Thread model: single
gcc version 4.5.1-tc32-1.3 (GCC)
```

github0null, 2022-10-12:
OK, i will fix it

github0null, 2022-10-12:
Uncheck this checkbox to disable: `-Wl,--gc-sections`

[image]

TheSnowfield, 2022-10-12:
> Uncheck this checkbox to disable: `-Wl,--gc-sections`
> 
> [image]

it does not completely work, 
then linker complains `tc32-elf-ld: unrecognized option '-Wl,-Map=./build/Debug/test.map'`.

github0null, 2022-10-12:
There are too many incompatible params, builder will insert some inherent params and we not support very-old gcc compilers

Maybe you need try a new project type:

[image]

In this toolchain type, you can choose linker type: `gcc` or `ld`，use `ld` and try build

**If the problem still exists, you need to switch to a new gcc version**

TheSnowfield, 2022-10-12:
thx, I will try it out.

TheSnowfield, 2022-10-12:
thanks! It works like a charm. :)
btw, how to add a soc-specific template to EIDE? I consider creating a PR.

github0null, 2022-10-12:
What is `soc-specific template` ?

If you means 'project template', you can share them in this repo: https://github.com/github0null/eide-templates

TheSnowfield, 2022-10-12:
okay. that's what I wondering. thanks!
closed.

closed 2022-10-12 by TheSnowfield