github0null/eide #181: Create project from a script

duaneellissd, 2022-10-26:
**What you want to do**
I'd love to (1) provide a list of directories, and (2) a list of 90% of the settings for a project, then (3) just ask a few questions and (4) generate a project that can be loaded.

This is sort of like what CMAKE does, but Cmake will not work for my purposes and I thought i would try em-ide

**Describe the problem**
Given:  A list of source directories (relative paths from some place)
Given: A list of include directories (again, relative paths)
Given: Various compiler options to use for all targets.

For each TARGET (cpu/board) 
Given:  A compiler (GNU [riscv and cortexM3, and Microblaze], and Keil)
Given:  A linker script
Given: Various Compiler command line options, specific to the project

My goal is to Generate/create: a project file for each of the targets.

I specifically do not want to use a "clickity-clickity" 30 minutes with a mouse dragging and dropping and adding, and tweaking - I want to script this process of creating a project.

**Expected behavior**

Some script based means to create a project file

** An example **

This example from TI gives an example of the problem;
https://github.com/lab11/cc2538-base/tree/master/foundation-firmware/1.0.1.0/driverlib/cc2538/examples
There are many example applications - most of these use IAR..

In my case - I have 4 different targets, 3 using GNU tools, 1 using KEIL
Looking at a simple way to create all the projects uniform ally
and not via 12 different 'mouse click sessions' that are slightly different on each one.

github0null, 2022-10-27:
You need write some script to generate `.eide/eide.json` and `.eide/xx.xx.options.xx.json`

`.eide/eide.json` is used to descript your project

github0null, 2022-10-27:
But i suggest that you can use `Import Project` function,

We support import `IAR/MDK/Eclipse` project informations

For more details, see: https://em-ide.com/docs/getting-started/import_prj

duaneellissd, 2022-10-27:
It will not import projects from the versions of Eclipse that i use.
For example:  
https://www.microsemi.com/document-portal/doc_download/137732-dg0472-running-webserver-and-tftp-server-on-smartfusion2-devices-using-iwip-and-freertos-libero-soc-v11-7-sp2-demo-guide

I don't have a Xilinx (microblaze) project handy but - i fully suspect that it will choke on that also.

And I wonder if it will work with STm32 CUBE example projects.

The issue is this:  (I believe) - every chip vendor creates customized versions of eclipse, it is so customized that they project files are weird and custom thus you will have a wonderful time trying to important every possible weird thing.

This is also why tools like CMAKE cannot create projects for these solutions.

github0null, 2022-10-28:
> every chip vendor creates customized versions of eclipse

Every eclipse c/c++ project have `.cproject` and `.project` files, these files provide some compatibility

So, this plug-in will parse `.cproject` and `.project` to import project informations

duaneellissd, 2022-10-28:
Again, this does not work - I tried this using the MicroSemi Eclipse, I get:

a) A long list of errors and warnings (see below)
b) When I try to compile - it compiles numerous files that ARE not part of the project.
Example: 
   The .cproject has a list of directories that *ARE* source folders and should be compiled
   Instead, the IDE program searches all subfolders and compiles all files that match: *.c
   The IDE also appears to *NOT* honor numerous CFLAGS specified in the .cproject
   Items such as: What are, and are not include directories

The directory has source code orginized into a series of subdirs.
Each subdir has a "src" and/or a "include" subdirectory
This includes for example:
  (a) a few "library like subdirectories" set of hardware drivers, and protocol handlers.
  (b) a "primary application" ([e: the function main()]- that uses the above "library like subdirectories"
  (c) a bootloader application {another 'main() function'} it uses a subset of the library like sub directories.
  (d)a "development board" test application [another main()] that only runs on the development board and various "library like subdirectories"

We have a different '.project' and '.cproject' for each of the above.
To work on a specific project, you copy in the correct: 'pair' of ".project/.cproject" files.
The key factor here is the MULITPLE apps share many of the subdirectories sort of like "code libraries"
But - each application provides a unique "config.h" file to be used when building that library.

An example of this is: "FreeRTOSConfig.h" - and "lwip_opts.h" which are stored in the specific applications include folder.  Thus when building application "foo" there is a "-I" path that points to the current applications include folder.

To work: "bootloader" - you copy in the bootloader ".project/.cproject" files and rebuild.
To work: "uint_testapp" - copy in the testapp: .project/.cproject files and rebuild
And so on for other apps

To try this out - I setup and verified that I could build and run one of the applications.
Then tried to import this application, i get this "spew" from the tool
I also tries to build the app (ignoring the warnings because its not clear how to resolve them)

=====
OUTPUT FROM THE IMPORT STEP FOLLOWS
=====
!!! Warning !!!

There are some incompatible arguments between 'eclipse' and 'eide', you need check and add them to eide project manually !

When you have solved these incompatible problems, you need delete this note file, otherwise this file will be show again !

---

##### Configurations For All Targets #####

//
///// Target: 'Debug' /////
//

Incompatible Args:
  /:
    globalArgs:
      - <Create flash image> = true
      - <Create extended listing> = undefined
      - <Print size> = true
      - <Optimization Level> = ilg.gnuarmeclipse.managedbuild.cross.option.optimization.level.none
      - <Debug level> = ilg.gnuarmeclipse.managedbuild.cross.option.debugging.level.max
      - <Debug format> = undefined
      - <undefined> = GNU Tools for ARM Embedded Processors
      - <Architecture> = ilg.gnuarmeclipse.managedbuild.cross.option.architecture.arm
      - <ARM family (-mcpu)> = ilg.gnuarmeclipse.managedbuild.cross.option.arm.target.mcpu.cortex-m3
      - <Instruction set> = ilg.gnuarmeclipse.managedbuild.cross.option.arm.target.instructionset.thumb
      - <Prefix> = arm-none-eabi-
      - <C compiler> = gcc
      - <C++ compiler> = g++
      - <Archiver> = ar
      - <Hex/Bin converter> = objcopy
      - <Listing generator> = objdump
      - <Size command> = size
[... 118 lines cut ...]
arm-none-eabi-gcc.exe: warning: '-x c' after last input file has no effect
arm-none-eabi-gcc.exe: fatal error: no input files
compilation terminated.
>> [  2%] CC 'app_campie/src/generated/sys_config.c'
arm-none-eabi-gcc.exe: warning: '-x c' after last input file has no effect
arm-none-eabi-gcc.exe: fatal error: no input files
compilation terminated.
>> [  2%] CC 'app_campie/src/generated/sys_config_SERDESIF_1.c'
arm-none-eabi-gcc.exe: warning: '-x c' after last input file has no effect
arm-none-eabi-gcc.exe: fatal error: no input files
compilation terminated.

** >> MY NOTES: Finally - a directory that has source it should compile! **

>> [  3%] CC 'app_campie/src/main.c'
arm-none-eabi-gcc.exe: warning: '-x c' after last input file has no effect
arm-none-eabi-gcc.exe: fatal error: no input files
compilation terminated.

 ERROR  compilation failed at : "c:\projects\wx_sat\wxsat\Software\BasePlatform\app_campie\src\main.c", exit code: 1

 ERROR  build failed !, elapsed time 0:0:0

 *  The terminal process "cmd.exe /C "unify_builder -p c:\projects\wx_sat\wxsat\Software\BasePlatform\build\Debug\builder.params"" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

github0null, 2022-10-29:
The import tool will only import basic project information, including: source files, containing directories, macro definitions, and a few common compile parameters

If your project is complex, you will need to manually adjust the imported project to the desired state.

The plug-in also imports eclipse's exclusion list and filters files that do not need to be compiled.

Please provide the file '.cproject/.project 'so that I can find out why the exclusion list is not work.

duaneellissd, 2022-10-29:
I recognize that my project is complex, and I have numerous others just like it -that I would love to transition to vscode, however - it would be far simpler if I could create projects via a script instead.

GITHUB rejects the filenames: ".cproject" and ".project" - so Instead I renamed them to '.txt' files

NOTE: I am not using exclusion lists, I just list the directories where the source located.
(You'll see these in the <sourceEntires> element in the cproject file).

[cproject.txt](https://github.com/github0null/eide/files/9894684/cproject.txt)
[project.txt](https://github.com/github0null/eide/files/9894686/project.txt)

github0null, 2022-10-30:
> -that I would love to transition to vscode, however - it would be far simpler if I could create projects via a script instead.

Scripts can be written on their own because everyone has different needs, just generate a few files according to the rules, and you're done

https://github.com/github0null/eide/issues/181#issuecomment-1292953139