github0null/eide #182: View compiler command line

duaneellissd, 2022-10-26:
**What you want to do**
How do I view the entire compiler command line that is being used?

**Describe the problem**
I downloaded an example project, it found my Keil tools and builds -yea, nice experience out of the box.
However - started looking at the project and I see these two settings:
```
        "EIDE.SourceTree.AutoSearchIncludePath": true,
        "EIDE.SourceTree.AutoSearchObjFile": true,
```
My hunch is - this tool goes and adds a "-I" path for every location it finds a .h file
For me - this is not going to work, I need to control the specific order and need to exclude directories from the search path, example: I have different include directories depending upon the tool or target. 

**Expected behavior**
How can I control the include paths? and search order in a project.
And - I need to control which directories are compiled, for example - I have mixed source code in the project, some C files are for STM32 - and some are for RiscV - the tool seems to compile "everything-*.c" and I need to only compile very specific directories, not every possible directory that can be found.

github0null, 2022-10-27:
We support 2 kind of source folder type

For project resource function, we have some docs:

https://em-ide.com/docs/modules/project_resource

github0null, 2022-10-27:
For workspace settings, these settings are used to compatible with old project

```
EIDE.SourceTree.AutoSearchIncludePath
EIDE.SourceTree.AutoSearchObjFile
```

https://em-ide.com/docs/notice/compatibility_changes

duaneellissd, 2022-10-27:
ok my assumption was wrong.

how do i specify an include directory and a source directory?

github0null, 2022-10-27:
Please read docs before startup your project.

https://em-ide.com/docs

How to add include path:

https://em-ide.com/docs/modules/project_attr

closed 2022-10-28 by github0null