github0null/eide #188: 源码编译与打包失败

Joffygo, 2022-11-15:
在分别git clone了eide与node-utility的code之后，我进入eide的项目目录，用npm install安装了所需要的依赖，但是执行compile和webpack阶段一直失败。考虑到可能是依赖的版本问题，我自己试着改了一下一些依赖的版本，我发现，我所依赖的@types/vscode的版本号是1.73.0，但是其对应的tsc似乎是4.0以上的版本（与devdependencies里typescript@^3.7.3不匹配），于是我把@types/vscode降至1.67.0后，发现编译错误的数量确实大大下降（从46errors变为4个errors），但还是会报can not find name "document"。
[image]
另外，我又试了试webpack，发现我在最新版的nodejs环境（18.11.0）下会报错error:0308010C:digital envelope routines::unsupported。后来我把nodejs的版本降至16.18.0，虽然这个错误不报了，但是还报以下的错误
Module not found: Error: Can't resolve './Telemetry/TelemetryTask' in '/home/usr_name/eide/eide/src'
 @ ./src/extension.ts

我猜想，是否是我的的依赖或者环境跟项目所需要不符，大佬能提供下各个依赖准确的版本号吗（以及nodejs的版本号）？

github0null, 2022-11-15:
NodeJs 主版本用 14 的

库报错，就安装最低有效版本，去 package.json 里面看，

因为有些库更新了有不兼容的语法

https://discuss.em-ide.com/d/424/4

Joffygo, 2022-11-15:
感谢你的回答，不过事实上，我在发这个贴前就已经试过用npm安装package.json里面最低版本的依赖。但是依然会报错如下。
[image]
我试了很多次，发现把@types/vscode的版本变为1.67.0报的错误是最少的，只有一个黄字的error。如下
[image]
这个黄字的error是可以忽略是吗？因为我看把这个extension运行起来似乎也未曾报什么错。

github0null, 2022-11-15:
`devDependencies` 里的也要算

[image]

警告不用管

Joffygo, 2022-11-15:
好的哦，十分感谢^ ^

closed 2022-11-15 by Joffygo