github0null/eide #195: Support for other compilers

elabdess, 2022-12-13:
Is it possible to use eide with other eclipse-based projects like Renesas CCRX/GCC RX projects, (e2studio)?

or how to add support for another toolchain and parse this type of .cproject file.

I can provide any information and project example.

[test_project.zip](https://github.com/github0null/eide/files/10217516/test_project.zip)

github0null, 2022-12-14:
1. You need to use `import project` function, and select `Eclipse`, then import .cproject file
  docs: https://em-ide.com/docs/getting-started/import_prj#eclipse-project

2. Open vscode workspace file: `xxx.code-workspace` file, 
  add `EIDE.Toolchain.AnyGcc.InstallDirectory` and `EIDE.Toolchain.AnyGcc.ToolPrefix` setting items to setup **Your GCC Compiler** for this project
  [image]

elabdess, 2022-12-15:
Thanks for answering!

That helped, I was missing the prefix settings for the GCC based compiler, now it compiles . What about CCRX? it's not GCC based, is ti possible to use as base other toolchain configuration?

bellow is the link to the compiler manual for further information.

https://community-ja.renesas.com/cfs-file/__key/communityserver-discussions-components-files/174/CC_2D00_RX-Compiler-User_2700_s-Manual.pdf

Thanks a lot and great plugin!

github0null, 2022-12-16:
we not support `CCRX`

closed 2022-12-19 by github0null