github0null/eide #196: [question] How to Run extennsion for IDE in vscode

BojunZhong, 2022-12-19:
I tryed to press F5,But it shows Error
[image]

github0null, 2022-12-19:
I can't understand your means.

Please use 'clear and concise description', Tell me what you want to do, and what's happened

BojunZhong, 2022-12-19:
I want to Run EIDE extensions in Vscode.
[image]

After I click Run And Debug. It will show that
[image]

BojunZhong, 2022-12-19:
My purpose is not to use EIDE(Just like download from Market). My purpose is to learn its code, how is that works, and Run it.

github0null, 2022-12-19:
If you want to Debug this extension, you need to do some prepare works, please follow these steps:

- Install NodeJS 14
- Full clone this repo
- Run `npm install` in workspace install all dependences
- Press `ctrl+shift+b`, select `webpack dev` to build extension

After build done, then press F5 start debugger

BojunZhong, 2022-12-19:
我用的版本是node v14.15.0
前面3步都ok：
Install NodeJS 14
Full clone this repo
Run npm install in workspace install all dependences
到了最后一步
[image]

点击这个npm：webpack
开始报错
[image]

github0null, 2022-12-19:
- 确保你 clone 了最新的 master 分支，还有其所有的 submodules
- 关于如何开发，调试扩展，请参考官方的教程

closed 2022-12-20 by github0null