github0null/eide #221: [bug] RISC-V 工具链编译时，到Linker的步骤仍然使用-gcc.exe 调用命令。

lekia, 2023-02-28:
**What are you doing?**
正在把CodeBlock的编译选项移植到EIDE中。

新手，如有低级错误请见谅。

**Describe the bug**
使用了自定义的工具链 RV32-Toolchain，其中包含 riscv32-elf-gcc.exe 和 riscv32-elf-ld.exe。
在Linker选项里设置了Other linker options (--gc-section 和 -Map=).
编译完成后，CodeBlock 内会用 Other linker options 里的参数调用 riscv32-elf-ld.exe，无报错。
而使用EIDE，会用 Other linker options 里的参数调用 riscv32-elf-gcc.exe，报错未知的参数。
因为只能设置工具链前缀，而不能像CodeBlock里那样单独设置链接器，所以无法继续。

**To Reproduce**
Steps to reproduce the behavior:
1. 设置 RISC-V编译器为 CodeBlock 的 RISC-V 工具链。
3. 设置  Other linker options。
4. 编译成功。
5. 链接步骤报错，未知参数。

**Expected behavior**
编译完成后，调用 riscv32-elf-ld.exe 而不是 riscv32-elf-gcc.exe。

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11 10.0.22000
 - EIDE Version v3.11.1
 - VSCode Version v1.69.0 insider

**Additional context**

github0null, 2023-02-28:
gcc 会自动调用 ld 的

用 gcc 而不用 ld 的好处是参数有更多的兼容性，而且 gcc 会自动加一些常用的参数（比如 -lc ...），用 ld 要自己写很多参数

目前只有 `Any GCC` 类型的项目能选 linker 的类型（为了兼容一些老的工具链），其它的都是直接用 gcc

---

有些选项是由勾选框的：

--gc-section 选项：
[image]

有些选项 eide 已经自动生成了

比如 -Map=

---

使用 `Show Compiler CommandLine` 可以直接查看生成的固定参数：

[image]

github0null, 2023-02-28:
如果要通过 gcc 传递参数给 ld，使用：

`-Wl,你的参数` 即可

lekia, 2023-02-28:
感谢！！
顺带请教一下，undefined reference to `asm' 如何解决？之前一直使用STM32和C51，asm() 是可以正常使用的，现在这个项目 codeblock 也可以正常编译，但是 vscode 编译，无论是使用 EIDE 自带的，还是 codeblock 的工具链，都会有 implicit declaration of function 'asm' 的警告，并且链接也会失败。

github0null, 2023-02-28:
gcc 中使用

`__asm`

lekia, 2023-02-28:
十分感谢，编译成功了。

closed 2023-02-28 by lekia

github0null, 2023-02-28:
更正一下：

- 如果你语言标准选的 c99：那就只能用 __asm(”xxx“)

- 如果你选的 gnu99，就可以用 asm("xxx")

[image]