github0null/eide #222: [bug] 这两天更新后会提示检测不到dotnet runtime，

hiddenblue, 2023-03-12:
**What are you doing?**
Tell us what you were doing when the bug occurred.

**Describe the bug**
**按照弹出的微软dotnet脚本安装后或者还是提示没有dotnet runtime**

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
提示信息：
```bash
[Info] : Embedded IDE launch begin
[Info] : Checking .NET6 runtime ...
[Info] : Exec cmd: 'dotnet --list-runtimes'
Microsoft.AspNetCore.App 7.0.2 [/usr/lib/dotnet/shared/Microsoft.AspNetCore.App]
Microsoft.NETCore.App 7.0.2 [/usr/lib/dotnet/shared/Microsoft.NETCore.App]
[Error] : Can not match .NET6 runtime
Error: Can not match .NET6 runtime
	at /home/chase/.vscode/extensions/cl.eide-3.11.1/dist/extension.js:20:59487
	at Generator.next (<anonymous>)
	at /home/chase/.vscode/extensions/cl.eide-3.11.1/dist/extension.js:20:51102
	at new Promise (<anonymous>)
	at n (/home/chase/.vscode/extensions/cl.eide-3.11.1/dist/extension.js:20:50847)
	at /home/chase/.vscode/extensions/cl.eide-3.11.1/dist/extension.js:20:58687
	at /home/chase/.vscode/extensions/cl.eide-3.11.1/dist/extension.js:20:61549
	at Generator.next (<anonymous>)
	at a (/home/chase/.vscode/extensions/cl.eide-3.11.1/dist/extension.js:20:50904)
[Info] : Not found [.NET6 Runtime](https://dotnet.microsoft.com/en-us/download/dotnet/6.0) !
[Info] : Embedded IDE launch done
```

使用安装dotnet安装脚本提示
```bash
 $ sudo bash ./dotnet-install.sh         
dotnet-install: Note that the intended use of this script is for Continuous Integration (CI) scenarios, where:
dotnet-install: - The SDK needs to be installed without user interaction and without admin rights.
dotnet-install: - The SDK installation doesn't need to persist across multiple CI runs.
dotnet-install: To set up a development environment or to run apps, use installers rather than this script. Visit https://dotnet.microsoft.com/download to get the installer.

dotnet-install: .NET Core SDK with version '6.0.406' is already installed.

```

**Desktop (please complete the following information):**
**平台 Ubuntu22.10**
**vscode 1.76.1**
**EIDE v3.11.1**

**Additional context**
```bash
[2023/3/12 11:0:43 GMT+0800] : launched
[Hidden] : spawnSync /home/chase/.eide/bin/utils/stvp_tools/stvp_utils ENOENT
Error: spawnSync /home/chase/.eide/bin/utils/stvp_tools/stvp_utils ENOENT
	at Object.spawnSync (node:internal/child_process:1111:20)
	at spawnSync (node:child_process:834:24)
	at Object.execFileSync (node:child_process:880:15)
	at Object.func [as execFileSync] (node:electron/js2c/asar_bundle:5:1812)
	at D.loadStm8DevList (/home/chase/.vscode/extensions/cl.eide-3.11.1/dist/extension.js:7:17938)
	at EventEmitter.<anonymous> (/home/chase/.vscode/extensions/cl.eide-3.11.1/dist/extension.js:7:10043)
	at EventEmitter.emit (node:events:526:28)
	at a.emit (/home/chase/.vscode/extensions/cl.eide-3.11.1/dist/extension.js:1:5877)
	at /home/chase/.vscode/extensions/cl.eide-3.11.1/dist/extension.js:20:75687
	at Generator.next (<anonymous>)
	at a (/home/chase/.vscode/extensions/cl.eide-3.11.1/dist/extension.js:20:50904)
```

麻烦作者看看啦

github0null, 2023-03-12:
你的 dotnet 是 7，要求是 6，主版本号必须一致

Microsoft.NETCore.App 7.0.2 [/usr/lib/dotnet/shared/Microsoft.NETCore.App]

hiddenblue, 2023-03-12:
OK. I found that I follow the instruction [](https://learn.microsoft.com/zh-cn/dotnet/core/install/linux-ubuntu-2210), however I installed the dotnet runtime 7.0 for my ubuntu. I delete its associated files manually and reinstalled the dotnet runtime 6.0. It worked.  
`sudo apt-get install -y dotnet-runtime-6.0 `

closed 2023-03-12 by github0null