github0null/eide #223: [bug] 导入KEIL 5工程时报错

ckforce, 2023-03-14:
导入KEIL 5工程时报错。（原工程可以用keil正常编译调试）

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: windows 11
 - EIDE Version：v3.11.1
 - VSCode Version：v1.76.1

github0null, 2023-03-14:
请提交 `C:\Users\<用户名>\.eide\cl.eide.log` 日志文件

ckforce, 2023-03-15:
[cl.eide.log](https://github.com/github0null/eide/files/10976586/cl.eide.log)

github0null, 2023-03-15:
问题源：
https://github.com/github0null/eide/blob/27608349e5437d3cad063f4758eb1dfc05c734ae/src/KeilXmlParser.ts#L780-L782

应该是 uvprojx 文件里的数据格式跟预期的不一样，导致解析时 index 溢出，

请上传一下你要导入的 `*.uvprojx` 文件，以及 keil 版本

ckforce, 2023-03-15:
[image]

[ble_dis_freertos.txt](https://github.com/github0null/eide/files/10978108/ble_dis_freertos.txt)

github0null, 2023-03-16:
已修复：https://github.com/github0null/eide/commit/8855f0802db565a3979e67aaa698b4963d695882

closed 2023-03-16 by github0null