github0null/eide #227: 导入模板提示警告信息

zhang8083, 2023-03-26:
[image]
在比较旧的版本中制作的模板文件，插件更新后再导入提示警告信息
模板一般做好就不会动，但是EIDE版本升级比较频繁，如果升级了得需要重新做一次模板，会比较麻烦。
有几个疑问想请教下
1.请问是什么原因？
2.如果直接导入会存在什么风险？
3.对于经常要使用导入模板的场景，有什么最佳实践的建议？

github0null, 2023-03-26:
导出模板时会生成一个 hash，加到名字后面

[image]

导入时，会检查模板 zip 文件的 hash，来确认模板文件是否完整（或已损坏），如果和名字上的 hash 不一致，就会报这个警告

这个跟版本没关系，hash 的检验方法没改过

github0null, 2023-03-26:
很久以前的老版本的插件没有检查 hash 的功能，也不生成 hash，

因此如果新版插件导入老版插件生成的模板，会直接跳过 hash 的检查，不会报这个错

---

你可以去看这个仓库里的模板，有很多都是没有 hash 的，也是一样可用的

https://github.com/github0null/eide-templates

---

出现这个问题，一般就是模板生成完之后，你又打开压缩文件，手动修改了 zip 里的内容，导致 hash 变了

手动改过的模板，你可以直接删了模板名字里的 hash 就行了，这样插件就不会进行检查

zhang8083, 2023-03-29:
问题已经解决，感谢耐心解答！

closed 2023-03-29 by zhang8083