github0null/eide #232: Multiple errors in buld

vmwsree, 2023-04-03:
I'm running the project on my mac, but a lot of files missing errors are showing, majorly in utils.tz could you help me resolve this

Is there any way to resolve this issue?\n\nAny help would be greatly appreciated. Thank you!

github0null, 2023-04-04:
Did you followed these steps ?

https://github.com/github0null/eide#how-to-build-

And please show your error messages, i only develop this project on Windows pc.

vmwsree, 2023-04-04:
Yes, I have tried running it on Windows also following the steps provided here is the error log. 

https://gist.github.com/vmwsree/1e2dba74d1af6c4a6a3ce6dd2f480c94

github0null, 2023-04-04:
You need to clone submodules !!!

https://github.com/github0null/eide/tree/master/lib

vmwsree, 2023-04-04:
Thank you , I have send u an email to connect, please check

also 

`WARNING in ./src/utility.ts
Module not found: Error: Can't resolve './Private/GithubProxy' in '/Users/vmwsree/ASH/eide/src'
 @ ./src/utility.ts
 @ ./src/extension.ts

WARNING in ./src/extension.ts
Module not found: Error: Can't resolve './Private/TelemetryTask' in '/Users/vmwsree/ASH/eide/src'
 @ ./src/extension.ts
`
hope these warnings are expected

github0null, 2023-04-04:
Ignore these warnings, they are inessential

```
Can't resolve './Private/GithubProxy' 
Can't resolve './Private/TelemetryTask'
```

closed 2023-04-06 by vmwsree